tpaviot/pythonocc-core #831: Multiple module import failures

GuillaumeQr, 2020-05-11:
Hello, I'm starting to use pythonocc linked with IfcopenShell but for now I've not been able to use it. I've installed the version 7.4.0beta2 according to: https://github.com/tpaviot/pythonocc-core/issues/676#issuecomment-569490933

Most of the imports of OCC fail:
OCC.TopoDS
OCC.V3d
OCC.AIS
...
For the modules in OCC.Core I get the error message:
`No module named 'OCC."module name"'`

And for others such as SimpleGUI (OCC.Display.SimpleGUI) I get the error:
`DLL load failed: The specified module could not be found.`

I don't understand how to fix them.

tpaviot, 2020-05-11:
Move ```from OCC.TopoDS``` to ```from OCC.Core.TopoDS```, same for all OCC modules.

GuillaumeQr, 2020-05-11:
I've already tried but it doesn't work. I automatically get the error:
`DLL load failed: The specified module could not be found.`
Imports: `import OCC.Core`, `import OCC.Display`, work
Imports: `import OCC.Core.module`, `import OCC.Display.module`, don't work

GuillaumeQr, 2020-05-13:
I haven't understood the problem but I've fixed it in reinstalling pythonocc with the conda command:
`conda install -c dlr-sc pythonocc-core=7.4.0`
I was also working with IfcOpenShell. I had also to remove and to reinstall it after pythonocc with the command:
`conda install -c conda-forge -c oce -c dlr-sc -c ifcopenshell ifcopenshell`

closed 2020-05-13 by GuillaumeQr