tpaviot/pythonocc-core #833: Small display while using multiple tabs

arbrasington, 2020-05-12:
I am using an occ widget on a tab that is initially disabled. When i enable the tab and click on it, the widget appears, however it is not the full size like below. I currently have a fix that requires resizing the window, but I want the widget to automatically resize. I have tried using the resize, MustBeResized, and reset views within the pythonocc-core documentation but it doesn't seem to have any effect. The only way I have gotten it to resize on initialization is by setting it as the current tab from the start, but that will not work for future development because it needs to be the second tab opened.

Here is the widget class that I have created.

    class occWidget(qtViewer3d):

        def __init__(self, parent=None):
            super().__init__(parent)
            if parent is None:
                return

            self.InitDriver()
            self._display.display_triedron()
            self._display.set_bg_gradient_color([206, 215, 222], [128, 128, 128])
            self.context: AIS_InteractiveContext = self._display.GetContext()

            self.myQMenuBar = QtWidgets.QMenuBar(self)
            self.fileMenu = self.myQMenuBar.addMenu('File')
            self.editMenu = self.myQMenuBar.addMenu('Edit')
            self.importAction = QtWidgets.QAction('Import Geometry', self)
            self.importAction.triggered.connect(self.file_open)
            self.fileMenu.addAction(self.importAction)

        def InitDriver(self):
            winid = int(self.winId())  # assume PyQt5 Window handles
            self._display = OCCViewer.Viewer3d(window_handle=winid, parent=self)
            self._display.Create()
            self._display.SetModeShaded()
            self._inited = True

            # dict mapping keys to functions
            self._key_map = {ord('W'): self._display.SetModeWireFrame,
                             ord('S'): self._display.SetModeShaded,
                             ord('A'): self._display.EnableAntiAliasing,
                             ord('B'): self._display.DisableAntiAliasing,
                             ord('H'): self._display.SetModeHLR,
                             ord('F'): self._display.FitAll,
                             ord('G'): self._display.SetSelectionMode,
                             }
            self.createCursors()
            self.view = self._display.GetView()

       def resizeEvent(self, event):
            if self._inited:
                super().resizeEvent(event)
                self._display.OnResize()

        def file_open(self):

            options = QtWidgets.QFileDialog.Options()
            options |= QtWidgets.QFileDialog.DontUseNativeDialog
            file_name, _ = QtWidgets.QFileDialog.getOpenFileName(
                self,
                "Select CAD File",
                "",
                "All Files (*);;Python Files (*.py)",
                options=options)
            if file_name and 'igs' in file_name:
                print(file_name)
                shape = read_iges_file(file_name)
                self._display.DisplayShape(shape, update=True)
            else:
                msg = QtWidgets.QMessageBox()
                msg.setWindowTitle('File Type Error')
                msg.setText('Wrong file type. Choose "igs" file')
                msg.exec_()

[image]

tpaviot, 2020-06-16:
can you please share the full code so that I can run the script without additional work on my side. I may have a fix I'd like to test.

AlbertoCasetta, 2021-04-16:
I experienced the same problem.

Did you finally solved that?

Thank you in advance.

tpaviot, 2021-04-16:
Can you please have a look at the example https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_qt5_app.py I just added this demo a couple of days ago, it might help you solving this issue

AlbertoCasetta, 2021-04-16:
I looked to the example but in my case I start from another tab; when I resize the main window manually, If i go to the tab where OCC is displayed the size remained unchanged and to update the size of display I need to resize the main window setting, as active tab, the once which contains OCC.

Tanneguydv, 2021-04-16:
Have you tried to affect the qtViewer3d as an addwidget to a layout and then resize it to fit the layout?
I have simulate the problem here without resizing (see code below, I've commented the `resize` to get this bug) : 

[image]

```
self.canvas = qtDisplay.qtViewer3d(self)
self.ui.horizontalLayout.addWidget(self.canvas)
#self.canvas.resize(640, 480)
```

The resizing won't be dynamic with this simple code though

AlbertoCasetta, 2021-04-16:
Yes the qtViewer3D is assigned on a layout and the resizing works fine if I made the interactively resize (manually with the mouse) when the active tab on my application is the once which contains the qtViewer3D; if I do that from anothe tab it doesn't works.
I found a solution customizing the resizeEvent function of my application by dynamically switch on the tab which contains the qtViewer3D, resize the canvas based on the event resize and the switch back on the tab which was active when I made the resize event.

AlbertoCasetta, 2021-04-16:
At least is not really elegant but it works.

AlbertoCasetta, 2021-04-16:
Another possible solution is to create a custom slot function linked to the currentChanged signal of a tabview.
If the selected active tab is the once which contains the qtViewer3D the display is resized based on the frame size which contains the layout.