tpaviot/pythonocc-core #835: Downcasted AIS and BRepMesh_IncrementalMesh

e3dEF, 2020-05-13:
Hi,

first, I'm sorry I can't provide a MWE for the current release, because I couldn't figure out how to downcast an AIS to a shape. The code is for 0.18.1.

I'm wondering why I get different meshing results when exporting a shape and later exporting the shape downcast from AIS. Dumping the shapes used to mesh, I can see no difference. Do I have to reset something?

```
from OCC.Display.SimpleGui import init_display
from OCC.BRepPrimAPI import BRepPrimAPI_MakeCylinder
from OCC.BRepMesh import BRepMesh_IncrementalMesh
from OCC.StlAPI import StlAPI_Writer
from OCC.AIS import Handle_AIS_Shape_DownCast
from OCCUtils.Topology import dumpTopology

display, start_display, add_menu, add_function_to_menu = init_display()
shape = BRepPrimAPI_MakeCylinder(10., 20.).Shape()

def stl(shape, name):
    BRepMesh_IncrementalMesh(shape, 4.0, True, 4.0, True)
    stl_exporter = StlAPI_Writer()
    stl_exporter.Write(shape, name)

dumpTopology(shape)
stl(shape, 'A.stl')

ais = display.DisplayShape(shape, update=True)
ais_shape_handle = Handle_AIS_Shape_DownCast(ais)
ais_shape = ais_shape_handle.GetObject()
shape = ais_shape.Shape()

dumpTopology(shape)
stl(shape, 'B.stl')

# start_display()
```
A.stl:
[image]
B.stl:
[image]

Kind regards

rainman110, 2020-05-13:
Hi. The reason is not the Downcast but the DisplayShape command.

During DisplayShape, ```shape``` gets a new tesselation according to the Deflection parameter of the interactive context. Depending on this value, the shape is remeshed and the mesh is stored in the shape.

Therefore, you get a higher-res shape after display in the second stl export.

e3dEF, 2020-05-14:
Thank you for your reply! Asking myself, why it is not possible to change the triangulation using an incrementalMesh, even when Display is closed before retriangulation. Solved it by copying the shape without triangulation, though.

closed 2020-05-14 by e3dEF