tpaviot/pythonocc-core #836: OpenGL issue while creating simple GUI

GuillaumeQr, 2020-05-15:
Hi, I'm using the latest release of pythonocc and I'm trying to test it by running the test file: 
`test/core_display_wx_unittest.py`
When I run this script a window pops displaying the box but it closes automatically. On my terminal I get the error message:
```
INFO:OCC.Display.backend:backend loaded: wx
INFO:OCC.Display.SimpleGui:GUI backend set to: wx
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! OpenGL context reports version 4.5 but does not export required functions for 4.5 (glGetnCompressedTexImage)
Please report this issue to OpenGL driver vendor 'intel'
Display3d class successfully initialized.
```

I catch an error linked with OpenGL but I have no idea how to solve it. I tried to locate also the last call and this happens at the line 178 in OCCViewer.py: `self.Init(self._window_handle)`

tpaviot, 2020-06-16:
It's an issue related to your opengl driver. It may be an old graphic card, or an outdated opengl driver. Try to update the driver to its latest version.

cafhach, 2020-09-28:
We never managed to get OpenCASCADE visualization working with VNC - if that is the case for you too. What worked however was to use a VNC server which is using the X server of the host system.

waterbug, 2021-01-03:
Just to provide another data point:  I got the same error message about OpenGL context when using the pythonocc-core 7.4.1 package for python 3.7 from the conda-forge channel (build py37hc019675_0).  Then I checked what would happen with 7.4.0 by doing 'conda install pythonocc-core=7.4.0' from the dlr-sc channel -- which also installed freeimageplus-3.18.0 from the dlr-sc channel.  Modifying the code that instantiates OCCViewer.Viewer3d to pass in a handle (using GetHandle()), as required in 7.4.0,
it works fine.
Seems strange that 7.4.1 would encounter an OpenGL driver error and 7.4.0 does not ... that seems like it would be more than a minor version difference!

Further info about the system:
The laptop I'm running the code on is an HP ENVY m7 notebook with an Intel HD Graphics 620 and NVIDIA GeForce 940MX, and when I run "update driver" on each one, Windows claims that the best driver for each is already installed.

waterbug, 2021-01-03:
The above comment also applies using the python 3.8 conda packages from the conda-forge channel:  7.4.0 work fine, 7.4.1 gives the OpenGL driver error.

waterbug, 2021-01-04:
One more comment:  I am still getting the OpenGL error with the python 3.8, pythonocc-core 7.4.0 package from the conda-forge channel, but it runs fine (but 7.4.1 crashes) on Windows 10 *and* on Ubuntu 20 (actually on PopOS, which is derived from Ubuntu).

haifengas, 2021-01-25:
I have run it on version 7.4.1 and python3.8, but do not kown why it can.

`

        import sys
        from PyQt5.QtWidgets import QMainWindow, QApplication
        from OCC.Display.backend import load_backend
        from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
        
        class Main(QMainWindow):
            def __init__(self):
                super().__init__()
                load_backend("qt-pyqt5")
                from OCC.Display.qtDisplay import qtViewer3d
                self.canvas = qtViewer3d(self)
                self.canvas.resize(640, 480)
                self.canvas.InitDriver()
                display = self.canvas._display
                display.DisplayShape(
                    BRepPrimAPI_MakeBox(1, 1, 1).Shape(),
                    update=True
                )
                self.setCentralWidget(self.canvas)
                self.setWindowTitle("pythonOCC")
                self.resize(640, 480)
        
                self.show()
        
        if __name__ == "__main__":
            app = QApplication(sys.argv)
            win = Main()
            win.canvas.InitDriver()
            sys.exit(app.exec_())

`

tpaviot, 2021-04-01:
A possible fix that seems to work (at least it works on my machine). Open the file ``[...]\lib\site-packages\OCC\Display\qtDisplay.py``, and change the line ``38`` from:

```python
class qtBaseViewer(QtOpenGL.QGLWidget):
```
to
```python
class qtBaseViewer(QtWidgets.QWidget):
```
Any report is welcome

kozintsev, 2021-06-04:
I reproduce the problem on my notebook.

Notebook Dell G3 3500

GPU:
1. Intel UHD Graphics (Comet Lake-H GT2) [R1] [DELL]
2. NVIDIA GeForce GTX 1660 Ti (TU116M) [DELL]

C:\Users\okozi\.conda\envs\pyoccenv\python.exe C:/Users/okozi/Documents/GitHub/pythonocc-demos/examples/core_helloworld.py
qt-pyqt5 backend - Qt version 5.15.2
####### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGl_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! OpenGL context reports version 4.6 but does not export required functions for 4.5 (glGetnCompressedTexImage)
Please report this issue to OpenGL driver vendor 'intel'
Display3d class successfully initialized.
#########################################
OpenGl information:
  GLvendor: Intel
  GLdevice: Intel(R) UHD Graphics
  GLversion: 4.6.0 - Build 26.20.100.7642
  GLversionOcct: 4.4
  GLSLversion: 4.60 - Build 26.20.100.7642
  Max texture size: 16384
  Max FBO dump size: 4096x16384
  Max combined texture units: 192
  Max MSAA samples: 16
  Viewport: 1023x767
  ResolutionRatio: 1
INFO:OCC.Display.qtDisplay:key: code 16777249 not mapped to any function

Process finished with exit code 0

Possibly a problem with the GeForce Experience

Tanneguydv, 2021-09-14:
> A possible fix that seems to work (at least it works on my machine). Open the file `[...]\lib\site-packages\OCC\Display\qtDisplay.py`, and change the line `38` from:
>
> ```python
> class [...]

I've recently change the graphic card associate with my `python.exe`  from pythonocc env, I've chosen my nvidia Quadro T2000 rather than my integrated Intel one and it works only with the modifications you have suggested, thank you.

kozintsev, 2021-09-30:
> A possible fix that seems to work (at least it works on my machine). Open the file `[...]\lib\site-packages\OCC\Display\qtDisplay.py`, and change the line `38` from:
>
> ```python
> class [...]

This helped also the settings that @Tanneguydv wrote about

waterbug, 2021-09-30:
That is interesting! In my application I have a customized viewer in 
which I copied some of the
code from the qtBaseViewer.  Changing its base class to QWidget in my 
code caused a crash when
"makeCurrent" was called because that's not a method of QWidget, but as 
the Qt docs point out
"makeCurrent" is usually not necessary and in fact when I commented out 
all the calls
to "makeCurrent" everything worked fine, so this change seems like it 
might be ok.

On 9/30/21 1:15 PM, Oleg Kozintsev wrote:

>     A possible fix that seems to work (at least it works on my
>     machine). Open the file
>     |[...]\lib\site-packages\OCC\Display\qtDisplay.py|, and change the
>     line |38| from:
>
>     class  qtBaseViewer(QtOpenGL.QGLWidget):
>
>     to
>
>     class  qtBaseViewer(QtWidgets.QWidget):
>
>     Any report is welcome
>
> This helped also the settings that @Tanneguydv 
> <https://github.com/Tanneguydv> wrote about
>

akhenaton2, 2022-04-27:
Hello,
I am trying to find this file. I figure out the path by looking on internet "C:\Users\serge\.conda\envs\pyoccenv\lib\site-packages\OCC\Display\qtDisplay.py". Serge has to be of course replaced by your name/name of user.

In the path "C:\Users\serge\.conda" I only have a text file, I can not go deeper in the folder.

EricMaoMao, 2022-07-27:
@akhenaton2 
You can try to use `conda env list` command in your terminal to locate your path of environment, and try to find the file "qtDisplay.py".

But the solution of [tpaviot](https://github.com/tpaviot) in [https://github.com/tpaviot/pythonocc-core/issues/836#issuecomment-811636414](url) can not make out in my code, which I assume might be the problem of GL driver's compatibility. I will try to downgrade the driver.

My GL info like [kozintsev](https://github.com/kozintsev) is:

```
GL vendor: Intel
GL device: Intel(R) UHD Graphics 630 （another is NVIDIA GeForce GTX 1060）
GL version: 4.6.0 - Build 30.0.101.1960
GL Shader version: 4.6.0 - Build 30.0.101.1960
```
Error occur when use code:
```
display, start_display, add_menu, add_function_to_menu = init_display("qt-pyqt5")
display.DisplayShape(shapes, update=True)
```
with pythonocc-core-7.4.1 and occt-7.4.0.

Error message:
```
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! OpenGL context reports version 4.6 but does not export required functions for 4.5 (glGetnTexImage)
Please report this issue to OpenGL driver vendor 'intel'
```
in 
```
Traceback (most recent call last):
  File "main.py", line 21, in <module>
    display.DisplayShape(shapes, update=True)
  File "my-occ-env\lib\site-packages\OCC\Display\OCCViewer.py", line 469, in DisplayShape
    shape_to_display = AIS_Shape(shape)
  File "my-occ-env\lib\site-packages\OCC\Core\AIS.py", line 14005, in __init__
    _AIS.AIS_Shape_swiginit(self, _AIS.new_AIS_Shape(*args))
TypeError: in method 'new_AIS_Shape', argument 1 of type 'TopoDS_Shape const &'
```

jkbgbr, 2022-12-11:
Just solved this issue I had with both the cards on my laptop: an Intel UHD 620 and an Nvidia MX250. 
In both cases the solution was to do a clean install (removing the old and resetting to factory settings) of the latest driver downloaded from the vendor side, checking the driver through the OS did not yield anything.