tpaviot/pythonocc-core #838: SMESH and STLMESH module

godjenka, 2020-05-17:
Hi,

I have a set of triangles that I want to represent as a mesh on the pythonOcc GUI but apparently there are no SMESH and StlMesh classes anymore inside the OCC.Core. 

Does anybody know how I can go about it?

e-dub, 2020-07-02:
I do not use the newest version but in pythonOCC 0.18.1 you had to build SMESH first and then link it.  I do this as follows:

```
#SMESH
cd ~/opt 
git clone git://github.com/tpaviot/smesh.git
cd smesh
git checkout 6.7.5
mkdir build
cd build .. 
cmake ..                                                            \
-DBUILD_SHARED_LIBS=ON                        
make -j4
sudo make install

#pythonOCC
cd ~/opt 
git clone git://github.com/tpaviot/pythonocc-core.git
cd pythonocc-core
git checkout 0.18.1
mkdir build
cd build
cmake ..                                                            \
-DBUILD_SHARED_LIBS=ON                                              \
-DPYTHONOCC_WRAP_SMESH=ON                                           \
-DPYTHONOCC_INSTALL_DIRECTORY=/usr/lib/python3.8/dist-packages/OCC  \
-DPYTHON_EXECUTABLE=/usr/bin/python3.8                              \
-DPYTHON_INCLUDE_DIR=/usr/include/python3.8                         \
-DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.8.so 
make -j4
sudo make install
```

weiweiisok, 2021-11-26:
> I do not use the newest version but in pythonOCC 0.18.1 you had to build SMESH first and then link it. I do this as follows:
>
> ```
> #SMESH
> cd ~/opt
> git clone git://github.com/tpaviot/smesh.git
> cd [...]

Excuse me , How  can i use smesh module when i use high edition of pythonocc such as pythonocc 7.5.1?