tpaviot/pythonocc-core #840: Building pythonocc for geometry info and visualization only.

efirvida, 2020-05-18:
Dear all.
I want to use PythonOCC to create a small library for reading (step, iges, brep) CAD files, extract info as volume, centroid, inertia properties, bounding box..... and also export to json, and stl, file to render for the web. But I don't need any of the modeling tools.
Is there any advice to build it only with these features, in order to reduce package size, and build time?

tpaviot, 2020-06-02:
You have to compile pythonocc by your own. The CMakeLists.txt file enables to choose which packages to wrap. Modeling tools are mandatory, but you can set the ```PYTHONOCC_WRAP_OCAF``` and ```PYTHONOCC_WRAP_VISU``` to ```OFF``` (see https://github.com/tpaviot/pythonocc-core/blob/master/CMakeLists.txt#L92), that will result in a smaller package size, and you won't need opengl as well.

efirvida, 2020-06-02:
> You have to compile pythonocc by your own. The CMakeLists.txt file enables to choose which packages to wrap. Modeling tools are mandatory, but you can set the `PYTHONOCC_WRAP_OCAF` and [...]

And offscreen rendering will work with these functions disabled?

tpaviot, 2020-06-03:
No, offscreen rendering won't work, it requires opengl and, at least, a virtual frame buffer. But export to webgl will be possible (threejs, x3d), they only need a mesher

CsatiZoltan, 2020-06-03:
@efirvida If you release it as open source, could you post the link here? I would be interested.

efirvida, 2020-06-15:
> @efirvida If you release it as open-source, could you post the link here? I would be interested.

There is no problem with sharing the code. It's nothing from the other world. It is still under development but is functional:

CsatiZoltan, 2020-06-15:
@efirvida Thank you.

If you called the `to_png` method of `CADModel` from another function, could you prevent the graphics from being closed? This is related to my question #855.

efirvida, 2020-06-15:
> @efirvida Thank you.
> 
> If you called the `to_png` method of `CADModel` from another function, could you prevent the graphics from being closed? This is related to my question #855.

I made this only to save screenshoot of the picture, on my project I don need the window, so to run it well I also use **"PYTHONOCC_OFFSCREEN_RENDERER=1"** as an environment variable.
but for your example I think you just need to call **start_display()** after **display.DisplayShape(...**

im44pos, 2023-03-20:
> > @efirvida If you release it as open-source, could you post the link here? I would be interested.
> 
> There is no problem with sharing the code. It's nothing from the other world. It is still under development but is functional:
> 
> ```python
> import uuid
> import enum
> from subprocess import check_output
> 
> # OCC IMPORT
> from OCC.Core.BRepTools import breptools_Read
> from OCC.Core.BRep import BRep_Builder
> from OCC.Core.TopoDS import TopoDS_Shape
> 
> from OCC.Core.BRepGProp import brepgprop_VolumeProperties
> from OCC.Core.BRepMesh import BRepMesh_IncrementalMesh
> from OCC.Core.BRepBndLib import brepbndlib_Add
> from OCC.Core.Bnd import Bnd_Box
> from OCC.Core.GProp import GProp_GProps
> from OCC.Extend.DataExchange import read_stl_file, read_step_file, read_iges_file
> 
> from OCC.Core.Tesselator import ShapeTesselator
> from OCC.Display.WebGl import x3dom_renderer, threejs_renderer
> from OCC.Core.Quantity import Quantity_Color, Quantity_TOC_RGB
> from OCC.Display.SimpleGui import init_display
> 
> class CADFormats(enum.Enum):
>     CAD = ["step", "stp", "iges", "igs", "brep"]
> 
> # READERS
> class Readers(ABC):
>     @abstractmethod
>     def __call__(self, file: str):
>         ...
> 
> 
> class StepReader(Readers):
>     def __call__(self, file):
>         return read_step_file(file)
> 
> 
> class IgesReader(Readers):
>     def __call__(self, file):
>         return read_iges_file(file)
> 
> 
> class BrepReader(Readers):
>     def __call__(self, file):
>         shape = TopoDS_Shape()
>         builder = BRep_Builder()
>         breptools_Read(shape, file, builder)
>         return shape
> 
> 
> CAD_READERS = {
>     "brep": BrepReader,
>     "step": StepReader,
>     "stp": StepReader,
>     "iges": IgesReader,
>     "igs": IgesReader,
> }
> 
> 
> class Model:
>     def __init__(self, file):
>         self.file = file
>         self.model = None
> 
>     def __call__(self):
>         return self.read()
[... 136 lines cut ...]
>             f"\tX: {info['center_of_mass'][0]}",
>             f"\tY: {info['center_of_mass'][1]}",
>             f"\tZ: {info['center_of_mass'][2]}",
>         )
>         matrix_of_inertia = (
>             f"\t{info['matrix_of_inertia'][0]}",
>             f"\t{info['matrix_of_inertia'][1]}",
>             f"\t{info['matrix_of_inertia'][2]}",
>         )
>         bounding_box = (
>             f"\tMIN: {info['bounding_box'][0]}",
>             f"\tMAX: {info['bounding_box'][1]}",
>         )
> 
>         _str = (
>             f"Volume: {info['volume']}",
>             f"Size: {info['shape_size']}",
>             "Center of mass:\n" + "\n".join(center_of_mass),
>             "Matrix of inertia:\n" + "\n".join(matrix_of_inertia),
>             "Bounding box:\n" + "\n".join(bounding_box),
>         )
>         return "\n".join(_str)
> ```

I get an "NameError: name 'ABC' is not defined" on "class Readers(ABC):"

Tanneguydv, 2023-03-20:
add the following line to import :
`from abc import ABC, abstractmethod`

im44pos, 2023-03-20:
> add the following line to import : `from abc import ABC, abstractmethod`

@Tanneguydv Thank you.