tpaviot/pythonocc-core #842: DataExchange refactor and X3D export

tpaviot, 2020-05-22:
The DataExchange.py module became too big. Current work it to split the content to dedicated modules.

New modules XDE.py and X3D.py will contain the recent work related to X3D export.

see commit 009ea2b561c8857cffce2063a2ce5c54be4cb7d6

ping @andreasplesch

andreasplesch, 2020-05-23:
Looks good to me.

andreasplesch, 2020-05-27:
I found a few small oversights in the port which need a self._ .
Also I think it is a good idea to add a get_scene and a get_internalFaceEntries method.
I can provide PR#s and a working notebook soon.
Is this issue a good place to continue ?

andreasplesch, 2020-05-27:
started a PR: #844

andreasplesch, 2020-05-27:
Here is a notebook which demonstrates how to use the module. It then uses the x3d function to produce x3dom compatible HTML which the notebook can display.

https://nbviewer.jupyter.org/github/andreasplesch/OCCToX3D/blob/275a01e047db69108040ef8db3501d945e7a2737/notebooks/XDEscenegraph.ipynb

andreasplesch, 2020-05-28:
#845
see
https://nbviewer.jupyter.org/github/andreasplesch/OCCToX3D/blob/6a14e8a0dfeeb5d24a504f1ebfbd8c00fd662344/notebooks/XDEscenegraph2.ipynb
https://github.com/andreasplesch/OCCToX3D/blob/notebooks/notebooks/XDEscenegraph2.ipynb

tpaviot, 2020-05-29:
I included the officiel x3d.py module as a dependency, into the x3d_official subpackage:
```
import OCC.Extend.DataExchange.x3d_standard.x3d as XX3D
```
see commit be5d4ddc88ed370cffa653138c377898df5134a3

andreasplesch, 2020-05-29:
I think this is safe. I will let Don as the maintainer know.

tpaviot, 2020-06-09:
I pushed a set of commits with a very first try for the new X3D.py, base on the pure python Tesselator and the x3d standard package see 56d5bacf8b06432fcfee383a6f90b4bd39d8eeda

There's still a big work to do: testing, clean ups (some old code has not been removed yet), small issues with the official x3d package (creaseAngle is not a property of the IndexedTriangleSet but I used to work with it, I don't know if I have to manually add this property or use another class, for instance the IndexedFaceSet)

tpaviot, 2020-06-11:
@andreasplesch I face an issue while trying to use a ```Normal``` node:
The single line
```python
Normal(vector=[[1,0,0], [0,1,0]])
```
raises the following issue
```bash
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/thomas/Devel/pythonocc-core/src/Extend/DataExchange/x3d_standard/x3d.py", line 35739, in __init__
    self.vector = vector
  File "/home/thomas/Devel/pythonocc-core/src/Extend/DataExchange/x3d_standard/x3d.py", line 35749, in vector
    assertGreaterThanEquals('vector', vector, -1)
  File "/home/thomas/Devel/pythonocc-core/src/Extend/DataExchange/x3d_standard/x3d.py", line 1652, in assertGreaterThanEquals
    assert isGreaterThanEquals(value, minimum), fieldName + '=' + str(value) + ' fails assertGreaterThanEquals minimum=' + str(minimum)
  File "/home/thomas/Devel/pythonocc-core/src/Extend/DataExchange/x3d_standard/x3d.py", line 1641, in isGreaterThanEquals
    if each < minimum:
TypeError: '<' not supported between instances of 'list' and 'int'
```

There's a quality check I don't understand inthe ```Normal``` class definition (from the official x3d.py file):
```python
    def vector(self, vector):
        if  vector is None:
            vector = MFVec3f.DEFAULT_VALUE(self)
        assertValidMFVec3f(vector)
        assertGreaterThanEquals('vector', vector, -1)
        assertLessThanEquals('vector', vector, 1)
        self.__vector = vector
```
I don't understant the ```assertGreaterThanEquals``` and ```assertLessThanEquals``` purpose. What do I miss ?

andreasplesch, 2020-06-11:
MFVec3f is a list of tuples. So try

Normal( vector=[(1,0,0), (0,1,0)] )

But for values like (2,0,0) there is still an assertionError, working as
designed.

https://www.web3d.org/documents/specifications/19775-1/V3.3/Part01/components/rendering.html#Normal

says that normals need to be normalized. So, this is a minimal check for
that.

-Andreas