tpaviot/pythonocc-core #843: Export additional object information to STEP

dmh126, 2020-05-26:
I managed to successfully build a STEP file with multiple entities grouped into compounds using StepControl_Writer and I'm wondering how to add additional information that is associated to every entity. I'm interested mostly in object names and properties. 
In my STEP file every object is named like:
```
--Open CASCADE STEP Translator 7.4.1
    --Open CASCADE STEP Translator 7.4.1.1_1
       --mesh_name_undefined
    --Open CASCADE STEP Translator 7.4.1.1_2
       --mesh_name_undefined
```
while I'm trying to visualize them, and it comes from:
```
#7 = PRODUCT('Open CASCADE STEP translator 7.4 1', 'Open CASCADE STEP translator 7.4 1','',(#8));
```

I'd like to add unique names to products and for each of them add a set of properties similar to IFC. Unfortunately can't find any example and not sure if this is even possible.

A small piece of my code just to show how I tackle geometry:
```
step_writer = STEPControl_Writer()
builder = BRep_Builder()
root_compound = TopoDS_Compound()
builder.MakeCompound(root_compound)

for geometry in geometries:
    sew = BRepBuilderAPI_Sewing()
    vertices = np.array([gp_Pnt(p[0],p[1],p[2]) for p in geometry['vertices']])
    for face in geometry['faces']:
        sew.Add(create_face_from_points(vertices[face]))
    sew.Perform()
    builder.Add(root_compound, sew.SewedShape())

step_writer.Transfer(root_compound, STEPControl_AsIs)
step_writer.Write('filename.stp')
```

Krande, 2020-05-29:
Hi,

you might find issue https://github.com/tpaviot/pythonocc-core/issues/482#issuecomment-582920250 useful. In the example I add name to each part by using (at least I think it was) 

` Interface_Static_SetCVal('write.step.product.name', "yourpartname") `

Nice job with the MakeCompound to create the tree-structure. That was something I hadn't cracked yet :)

If you are able to share a short end to end example demonstrating the tree structure + name of entity written to a step file, I think that would be an awesome addition to the pythonOCC examples!

Best Regards
Kristoffer

tpaviot, 2020-06-02:
You have to use the OCAF framework https://www.opencascade.com/doc/occt-7.4.0/overview/html/occt_user_guides__ocaf.html There's an example for impor but not for export (see pythonocc-demos repository)

yli-cai, 2025-03-26:
> Hi,
> 
> you might find issue [#482 (comment)](https://github.com/tpaviot/pythonocc-core/issues/482#issuecomment-582920250) useful. In the example I add name to each part by using (at least I think it was)
> 
> `Interface_Static_SetCVal('write.step.product.name', "yourpartname")`
> 
> Nice job with the MakeCompound to create the tree-structure. That was something I hadn't cracked yet :)
> 
> If you are able to share a short end to end example demonstrating the tree structure + name of entity written to a step file, I think that would be an awesome addition to the pythonOCC examples!
> 
> Best Regards Kristoffer

Dear Krande, 

Thanks for your provided code example showing how to save the assembly file, your work has helped me a lot. 

I would like to ask you for some help about this topic. Here is my code snippet.

```
# pythonocc issue forum (#482 cite from Krande)
def export_component_A_with_component_Bs(
    file_name, component_A, component_B, application_protocol='AP214IS', progress_callback=None
):
    """
    Export the component_A with component_B as a STEP file.
    Returns True if successful, raises Exception on failure.
    """
    # get the component_A name and shape
    component_A_name, component_A_shape = component_A
    component_B_settings, component_B_shape = component_B

    # check if the shapes are valid
    if component_A_shape.IsNull():
        raise AssertionError('Component A shape is null.')

    if component_B_shape.IsNull():
        raise AssertionError('Component B shape is null.')

    if progress_callback:
        progress_callback(0, 'Starting STEP export...')

    if application_protocol not in ['AP203', 'AP214IS', 'AP242DIS']:
        raise AssertionError(
            f'application_protocol must be either AP203 or AP214IS. You passed {application_protocol}.'
        )

    # add .stp extension if not present
    if not file_name.lower().endswith('.stp'):
        file_name += '.stp'

    if os.path.isfile(file_name):
        print(f'Warning: {file_name} file already exists and will be replaced')

    assembly_mode = 1
    writer = STEPControl_Writer()
    fp = writer.WS().TransferWriter().FinderProcess()
    Interface_Static.SetCVal('write.step.schema', application_protocol)
    Interface_Static.SetCVal('write.step.unit', 'MM')
    Interface_Static.SetCVal('write.step.assembly', str(assembly_mode))

    # components
    components = [component_A_shape, component_B_shape]
    comp_names = [
        component_A_name,
        component_B_settings
    ]

    for i, comp in enumerate(components):
        Interface_Static.SetCVal('write.step.product.name', comp_names[i])
        status = writer.Transfer(comp, STEPControl_AsIs)
        if int(status) > int(IFSelect_RetError):
            raise Exception('Some Error occurred')

        item = stepconstruct.FindEntity(fp, comp)
        if not item:
            raise Exception('Item not found')

        item.SetName(TCollection_HAsciiString(comp_names[i]))

        if progress_callback:
            progress_callback(int(i / len(components) * 100), f'Exporting {comp_names[i]}...')

    # export the STEP file
    status = writer.Write(file_name)
    if int(status) > int(IFSelect_RetError):
        raise Exception('Something bad happened')

    if progress_callback:
        progress_callback(100, f'STEP file exported successfully to {file_name}')

    return True
```

In my application, I have a component A geometry, and at the same time, I get many component B shapes and component_B_shape is already `TopoDS_Compound`. However, what I want to achieve is to have an assembly file where stores two parts, one is component A and the other is component B shapes. In component B shapes, it will contain all the individual component B shape. My current code modified based on your example returns something like this:
```
Number of shapes at root : 2
Name : Component A
Name : Component B - (0.25, 0.10) 2
Name : Component B - (0.25, 0.10) 2.1
Name : Component B - (0.25, 0.10) 2.1.1
Name : Component B - (0.25, 0.10) 2.1.2
...
Name : Component B - (0.25, 0.10) 2.14.2
```
Can you give me some advice about how to form the structure like:
- root
  - Component A
  - Component B (0.25, 0.1)
    - component B (1)
    - component B (2)
    - component B (3)
    -  ....

It seems that you are super busy, so I would be very appreciated if you can help me out! 

Thanks for your reading and appreciated for your help in advance! 💓