tpaviot/pythonocc-core #846: Relations between entites

Hedgehogues, 2020-06-03:
Hello. I find to entities and relation between of python-occ is graph. There are some relations: https://www.opencascade.com/doc/occt-7.0.0/overview/html/occt__tutorial.html#OCCT_TUTORIAL_SUB2_3

I want to extract all relations as list of edges with types edges and vertexes like this:

```
[
   ((v0, type0), (v1, type1), edgeType0),
   ((v1, type0), (v0, type1), edgeType1),
   ....
]
```

I can to extract only vertexes like this:

```
from __future__ import print_function

from OCC.Core.IFSelect import IFSelect_RetDone
from OCC.Core.STEPControl import STEPControl_Reader
from OCC.Core.TopExp import TopExp_Explorer

from OCC.Core.TopoDS import topods_Face, topods_Edge

from OCC.Core.TopAbs import TopAbs_FACE

def get_faces(_shape):
    """ return the faces from `_shape`
    :param _shape: TopoDS_Shape, or a subclass like TopoDS_Solid
    :return: a list of faces found in `_shape`
    """
    topExp = TopExp_Explorer()
    topExp.Init(_shape, TopAbs_FACE)
    _faces = []

    while topExp.More():
        fc = topods_Face(topExp.Current())
        _faces.append(fc.Orientation())
        topExp.Next()

    return _faces

def main():
    filename = 'data/sample2/2 Plate 20.STEP'
    step_reader = STEPControl_Reader()
    status = step_reader.ReadFile(filename)
    assert status == IFSelect_RetDone
    number_of_roots = step_reader.NbRootsForTransfer()
    i = 1

    while i <= number_of_roots:
        ok = step_reader.TransferRoot(i)
        assert ok, 'Failed to find a suitable root for the STEP file'
        i += 1

    aResShape = step_reader.Shape()

    faces = get_faces(aResShape)
    print(faces)

if __name__ == '__main__':
    main()

```

How I can to extract edges and types?

tpaviot, 2020-06-03:
There's a builtin pythonocc extension that can traverse a shape topology and extract any kind of sub topology
```python
from OCC.Extend.TopologyUtils import TopologyExplorer

te = TopologyExplorer(myShape)

# loof over faces:
for face in te.faces():
    # do something

# loop over edges:
for edge in te.edges():
    # do something else

# same for te.wires()
```

Hedgehogues, 2020-06-03:
Okay. But vertex is face? What about other objects: https://www.opencascade.com/doc/occt-7.0.0/refman/html/class_topo_d_s___shape.html ?

Hedgehogues, 2020-06-05:
This method helps me

```
te = TopologyExplorer(aResShape)

for face in te.edges_from_face():
    pass
```

closed 2020-06-05 by Hedgehogues