tpaviot/pythonocc-core #847: Are Rational Bézier curves supported in PythonOCC-core?

krontzo, 2020-06-05:
I am constructing cubic Bézier curves using the control points (poles in OCC terminology).

But if I try to created a cubic rational Bézier curve, I get:

```
    _Geom.Geom_BezierCurve_swiginit(self, _Geom.new_Geom_BezierCurve(*args))

RuntimeError: Standard_ConstructionError

wrapper details:
  * symname: new_Geom_BezierCurve
  * wrapname: _wrap_new_Geom_BezierCurve__SWIG_1
  * fulldecl: Geom_BezierCurve::Geom_BezierCurve(TColgp_Array1OfPnt const &,TColStd_Array1OfReal const &)
```

The array of points and the array of reals have the same size. Even if all weights are 1, it keeps failling.

tpaviot, 2020-06-05:
see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_edge.py#L64 for an example of using ```Geom_BezierCurve```

krontzo, 2020-06-05:
That's the second thing I did, but it doesn't show weights.
The OpenCASCADE C++ reference says that `Geom_BezierCurve` can receive weights. I'll check again.

Thank you for your quick response.

krontzo, 2020-06-05:
From C++ api:

<https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_geom___bezier_curve.html>
<https://dev.opencascade.org/doc/refman/html/class_geom___bezier_curve.html>

`Geom_BezierCurve (const TColgp_Array1OfPnt &CurvePoles, const TColStd_Array1OfReal &PoleWeights)`

So I modified the example and it works for the 8-pole curve, but not for my 4-pole example. =(

```python
##Copyright 2009-2015 Thomas Paviot (tpaviot@gmail.com)
##
##This file is part of pythonOCC.
##
##pythonOCC is free software: you can redistribute it and/or modify
##it under the terms of the GNU Lesser General Public License as published by
##the Free Software Foundation, either version 3 of the License, or
##(at your option) any later version.
##
##pythonOCC is distributed in the hope that it will be useful,
##but WITHOUT ANY WARRANTY; without even the implied warranty of
##MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
##GNU Lesser General Public License for more details.
##
##You should have received a copy of the GNU Lesser General Public License
##along with pythonOCC.  If not, see <http://www.gnu.org/licenses/>.

import math

from OCC.Core.gp import gp_Pnt, gp_Lin, gp_Ax1, gp_Dir, gp_Elips, gp_Ax2
from OCC.Core.BRepBuilderAPI import (BRepBuilderAPI_MakeEdge,
                                     BRepBuilderAPI_MakeVertex)
from OCC.Core.TColgp import TColgp_Array1OfPnt
from OCC.Core.TColStd import TColStd_Array1OfReal
from OCC.Core.Geom import Geom_BezierCurve

from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

def edge(event=None):
    # The blud edge
    BlueEdge = BRepBuilderAPI_MakeEdge(gp_Pnt(-80, -50, -20),
                                       gp_Pnt(-30, -60, -60))
    V1 = BRepBuilderAPI_MakeVertex(gp_Pnt(-20, 10, -30))
    V2 = BRepBuilderAPI_MakeVertex(gp_Pnt(10, 7, -25))
    YellowEdge = BRepBuilderAPI_MakeEdge(V1.Vertex(), V2.Vertex())

    #The white edge
    line = gp_Lin(gp_Ax1(gp_Pnt(10, 10, 10), gp_Dir(1, 0, 0)))
    WhiteEdge = BRepBuilderAPI_MakeEdge(line, -20, 10)

    #The red edge
    Elips = gp_Elips(gp_Ax2(gp_Pnt(10, 0, 0), gp_Dir(1, 1, 1)), 60, 30)
    RedEdge = BRepBuilderAPI_MakeEdge(Elips, 0, math.pi/2)

    # The green edge and the both extreme vertex
    P1 = gp_Pnt(-15, 200, 10)
    P2 = gp_Pnt(5, 204, 0)
    P3 = gp_Pnt(15, 200, 0)
    P4 = gp_Pnt(-15, 20, 15)
    P5 = gp_Pnt(-5, 20, 0)
    P6 = gp_Pnt(15, 20, 0)
    P7 = gp_Pnt(24, 120, 0)
    P8 = gp_Pnt(-24, 120, 12.5)
    array = TColgp_Array1OfPnt(1, 8)
    array.SetValue(1, P1)
    array.SetValue(2, P2)
    array.SetValue(3, P3)
    array.SetValue(4, P4)
    array.SetValue(5, P5)
    array.SetValue(6, P6)
    array.SetValue(7, P7)
    array.SetValue(8, P8)
    weights = TColStd_Array1OfReal(array.Lower(), array.Upper())
    weights.SetValue(1, 1.0)
    weights.SetValue(2, 1.0)
    weights.SetValue(3, 1.0)
    weights.SetValue(4, 2.0)
    weights.SetValue(5, 2.0)
    weights.SetValue(6, 1.0)
    weights.SetValue(7, 1.0)
    weights.SetValue(8, 1.0)
    nonratcurve = Geom_BezierCurve(array)
    ratcurve = Geom_BezierCurve(array, weights)
    
    for curve in [nonratcurve, ratcurve]:
        ME = BRepBuilderAPI_MakeEdge(curve)
        GreenEdge = ME
        V3 = ME.Vertex1()
        V4 = ME.Vertex2()
        display.DisplayColoredShape(GreenEdge.Edge(), 'GREEN')
        display.DisplayShape(V3)
        display.DisplayShape(V4, update=True)

    display.DisplayColoredShape(BlueEdge.Edge(), 'BLUE')
    display.DisplayShape(V1.Vertex())
    display.DisplayShape(V2.Vertex())
    display.DisplayColoredShape(WhiteEdge.Edge(), 'WHITE')
    display.DisplayColoredShape(YellowEdge.Edge(), 'YELLOW')
    display.DisplayColoredShape(RedEdge.Edge(), 'RED')

if __name__ == '__main__':
    edge()
    start_display()
```

I'm going to close it since it works for your modifed example.

closed 2020-06-05 by krontzo