tpaviot/pythonocc-core #850: How to join splines?

CsatiZoltan, 2020-06-05:
B-splines can be [split](https://github.com/tpaviot/oce/search?q=SplitBSplineCurve&unscoped_q=SplitBSplineCurve), but didn't find any method to merge/join them. The reason I want this is because [GeomFill_BSplineCurves](https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_geom_fill___b_spline_curves.html#details) accepts at most 4 splines, but my boundary consists of more than that.

tpaviot, 2020-06-11:
I would first join edges into one wire, a convert the resulting wire into a bspline curve

CsatiZoltan, 2020-06-11:
In the meantime, I came to the same conclusion that I must create a wire. However, I didn't convert the wire to a `BSplineCurve`, as my goal is to create a surface out of it. So what I did:

```python
face_from_wire = BRepBuilderAPI_MakeFace(wire.Wire())
if not face_from_wire.IsDone():
    raise Exception('Face could not be created.')
```

What is strange is that the exception above was never raised in my code but still, in some cases (especially when the boundary `BSplineCurve`s are complicated) the face was not created, as I checked it with
```python
display.DisplayShape(face_from_wire.Face(), update=True)
```
What is its cause?

CsatiZoltan, 2020-06-13:
Anyway, the question in my previous post is not related to joining curves, so I close this issue.

closed 2020-06-13 by CsatiZoltan