tpaviot/pythonocc-core #852: Missing algorithm

hbadi, 2020-06-08:
Hi,

I cant't see the BRepOffsetAPI_Sewing class, is it a miss ?

Best regards.

tpaviot, 2020-06-08:
yes, right, it seems it's missing

see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_boolean_sewed_shapes.py, it should work using the last pythonocc release

hbadi, 2020-06-08:
Great, thanks.

hbadi, 2020-06-08:
Is there any equivalent to c++ dynamic cast operator TopoDS::Solid/Shell/... in python ?

tpaviot, 2020-06-08:
```topods_Solid```, ```topods_Shell``` etc.

tpaviot, 2020-06-15:
```BRepOffsetAPI_Sewing``` is just an alias for ```BRepBuilderAPI_Sewing```:

```c++
#include <BRepBuilderAPI_Sewing.hxx>
//! Sew the shapes along their common edges
typedef BRepBuilderAPI_Sewing BRepOffsetAPI_Sewing;

#endif // _BRepOffsetAPI_Sewing_HeaderFile
```
You can use the latter.

closed 2020-10-03 by tpaviot