tpaviot/pythonocc-core #854: dumping Graphic3d_BT_Depth does not dump the depth buffer

daniel1v, 2020-06-13:
I am trying to render depth images and therefore modified the "core_offscreen_rendering.py" example as follows:

`from OCC.Display.OCCViewer import Viewer3d
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.Graphic3d import Graphic3d_BT_Depth

offscreen_renderer = Viewer3d(None)
offscreen_renderer.Create()
offscreen_renderer.SetSize(1024, 768)
offscreen_renderer.SetModeShaded()
my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()
offscreen_renderer.DisplayShape(my_box, update=True)
data = offscreen_renderer.GetImageData(Graphic3d_BT_Depth)
offscreen_renderer.View.Dump('./capture_jpeg.jpeg', Graphic3d_BT_Depth)
`

The rendered image only contains the value 1 for every pixel. Is this a bug or am I doing something wrong?

Also: The exported image is of another size than the renderer: 1040x807. This happens also when using Graphic3d_BT_RGB.

tpaviot, 2020-06-16:
I have the same result than you get, on Linux the image generated using Graphic3d_BT_Depth buffer type is black. I'm afraid I can't help, I never studied this part of the opencascade code.

daniel1v, 2020-06-21:
okay, although, thrank you for your answer. i solved my issue by writing my ownn shader. now the values are as expected. maybe the default shader uses the depth buffer for sth different than depth.

pridi, 2021-03-21:
@daniel1v I have the same issue as you. Could you please help me with the shader? How did you solve the problem that the images are of different size than the renderer? The same is for me. Thank you

daniel1v, 2021-03-24:
I am sorry, I cannot remember, how I solved it exactly.

The core of the solution was to use a custom opengl renderer. Somewhere in the pythonocc-demos repo is a simple example for that.