tpaviot/pythonocc-core #855: Prevent the display being closed when a function returns

CsatiZoltan, 2020-06-13:
I separated my visualization functionality to a separate function, like this:

```python
from OCC.Display.SimpleGui import init_display

def core:
    # Set up a `TopoDS_Shape` object, called `shape`
    # ...
    # Call the plotting function
    plot_shape(shape)

def plot_shape(shape):
    display, start_display, add_menu, add_function_to_menu = init_display()
    display.DisplayShape(shape, update=True)
    input("Press Enter to continue...")
```
The last line is there to prevent the figure from being closed. However, this is not a good solution because
- the program termination is paused
- I cannot focus on the figure, so I cannot rotate, zoom, pan, etc.

efirvida, 2020-06-15:
try;
 
``` python
def plot_shape(shape):
    display, start_display, add_menu, add_function_to_menu = init_display()
    display.DisplayShape(shape, update=True)
    start_display()
```

CsatiZoltan, 2020-06-16:
@efirvida Thanks, that works. However, the execution is suspended until I keep the window open. Is there a way to make the display independent of the main program thread?