tpaviot/pythonocc-core #859: Measure volume step by step issue and stp solid split issue. Thanks for your helping!

a740615405, 2020-06-22:
I have a 3D model made by CATIA, and convert to stp file. Then I read it by pythonocc, and make a plane. I want to measure the volume below the plane. After this, I need to move the plane higher of 1mm (sometimes I need to rotate the plane), and measure the volume below the new plane again. Then I will repeat the above steps, until 100mm or more.

In this progress, I meet 2 big problems:

1. I try these two functions: BRepAlgoAPI_Cut() and BOPAlgo_Splitter() to split the model and then measure the remaining solid, but it is very very slow to finish (my model is a little complex). So I want to know if there is another fast method. And sometimes, my model can’t be split, this is the second problem.
2. Just like I said, sometimes, my model can’t be split at some height. And I use BRepAlgoAPI_Check() to check my model, the IsValid() attributes said False. But the model is really made in CATIA, and it can be split in CATIA, it can also be displayed in pythoncc! So I don’t know what’s wrong with my data, and how to avoid this.

Please help me, and I will be glad to receive your reply, thank you very much!

[image]

[my_stp_split.zip](https://github.com/tpaviot/pythonocc-core/files/4810787/my_stp_split.zip)

tpaviot, 2020-06-23:
1. What do you mean with "very slow" ? what are your requirements in terms of run time ? It could be worth moving to vtk to do that (mesh the shape, export the mesh to a vtkPolyData, and use builtin vtk classes to split / measure)

2. It's not because the model can be displayed that it is topologically consistent. And the fact that Catia was used to build the model it is not a warranty that the STEP file is ok.

a740615405, 2020-06-28:
> 1. What do you mean with "very slow" ? what are your requirements in terms of run time ? It could be worth moving to vtk to do that (mesh the shape, export the mesh to a vtkPolyData, and use builtin [...]

Thanks for your replying.
1. I tried the 3D model in zip file. I made a test of 400 'for' cycles, it will take about 25 minutes. I think if it completes in 2 or 3 minutes, that will be ok. 
2. As the 3D model designer said, all the models are made by CATIA with the same step, so they don't think there is any wrong in the model. And I tested some other model we made, all of them have the same problem. So I think there may be 2 directions: 1. How can I convert the model to a step file(or other files)by CATIA that will be identified by pythonocc? 2.If there is any method that can fix the model in pythonocc?
3. I have never used vtk, will the problem 2 be solved in vtk?