tpaviot/pythonocc-core #862: [use/howto] Installing pythonocc-core on Amazon Linux 2

imgeaslikok, 2020-06-30:
Does anyone know how to install pythonocc-core on Amazon Linux 2 for using it in a Django application which it's deployed to Elastic Beanstalk? I saw the build method with cmake, but when the instance is renewed in the Elastic Beanstalk environment with instance autoscaling, I will have to repeat this process continuously. So I think that method doesn't work for me. I would be glad if you help.

closed 2020-07-14 by imgeaslikok

tpaviot, 2020-07-14:
did you solve the issue ?

imgeaslikok, 2020-07-14:
yes sir. i solved according to the answer in this link: https://stackoverflow.com/questions/62865974/installing-anaconda-on-amazon-elastic-beanstalk-to-use-in-django-application

tpaviot, 2020-07-14:
great, thanks for sharing