tpaviot/pythonocc-core #866: Select.py a magic name?

Paul-Wray, 2020-07-08:
I am running through the examples and happened to name on of them as 'select.py'. I later tried to run another example from a different source file, and found (after much confusion) that select.py was actually being called up first. Is select.py a magic name for some mechanism in pythonocc?

tpaviot, 2020-07-08:
examples are available at https://github.com/tpaviot/pythonocc-demos/tree/master/examples, and I can't see any select.py. I can' remember such an example name, where did you find it ?

Paul-Wray, 2020-07-08:
I gave it that name, while running the code from core_display_callbacks.py.

Paul-Wray, 2020-07-08:
Another question if I may? I am trying to get a list of selected points from a point cloud. I nave not seen any examples like that.
The point cloud is successfully highlighted in the display with SHIFT-left click and drag.  I have created and attached an owner for the point cloud, and after selection is made I get back a TColStd_HPackedMapOfInteger, but I cant seem to access the contents.