tpaviot/pythonocc-core #867: Support for XCAFDoc_DimTolTool

dmontagu, 2020-07-11:
I'm looking to make use of the `GetGDTPresentations` method of the `XCAFDoc_DimTolTool`: http://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=src/XCAFDoc/XCAFDoc_DimTolTool.hxx;h=0107e59b125624e8580131a0834c533ab4660f74;hb=HEAD#l242

However, I noticed the class isn't currently wrapped:

https://github.com/tpaviot/pythonocc-core/blob/68c9e0c24939f98cfb971ab4dcc3ef5833da8052/src/SWIG_files/wrapper/XCAFDoc.i#L7979

Is there any specific reason why this class isn't currently wrapped? Is it just a matter of someone putting in the effort to wrap it, or is there something more problematic?

With a little direction about what needs to be added/changed, I'd be happy to try to put together a pull request adding support.

tpaviot, 2020-07-13:
Issue confirmed. ```XCAFDoc_DimTolTool``` is missing. The .i swig files are generated using the pythonocc-generator (https://github.com/tpaviot/pythonocc-generator), the missing class is explicitly excluded from the wrapper (see https://github.com/tpaviot/pythonocc-generator/blob/master/src/Modules.py#L913).

tpaviot, 2020-07-13:
I pushed a fix at https://github.com/tpaviot/pythonocc-core/pull/868

tpaviot, 2020-07-13:
Fixed.

closed 2020-07-13 by tpaviot

dmontagu, 2020-07-20:
Thank you so much! Appreciate the fast turnaround.