tpaviot/pythonocc-core #870: Compile error BOPAlgo_Tools TreatCompound

hoba87, 2020-07-14:
I am trying to compile the project, but getting an error: TreatCompound is not a member of BOPAlgo_Tools (see below for long message)

I have tried with:
pythonocc-core@7.4.0 and pythonocc-core@master
opencascade versions 7.4.0 / 7.4.0p1
swig 3.0.12 

My assumption is that the error is due to some refactoring, however I would have expected that any of the above versions should have worked. Can anybody help me to get pythonocc-core compile?

```
[ 49%] Building CXX object CMakeFiles/_BOPAlgo.dir/Unix/x86_64-Release-/BOPAlgoPYTHON_wrap.cxx.o
/home/common/software/python-pkgs/pythonocc-core/cmake-build/Unix/x86_64-Release-/BOPAlgoPYTHON_wrap.cxx: In function \u2018PyObject* _wrap_BOPAlgo_Tools_TreatCompound(PyObject*, PyObject*)\u2019:
/home/common/software/python-pkgs/pythonocc-core/cmake-build/Unix/x86_64-Release-/BOPAlgoPYTHON_wrap.cxx:10055:7: error: \u2018TreatCompound\u2019 is not a member of \u2018BOPAlgo_Tools\u2019
       BOPAlgo_Tools::TreatCompound((TopoDS_Shape const &)*arg1,*arg2,*arg3);
       ^~~~~~~~~~~~~
CMakeFiles/_BOPAlgo.dir/build.make:82: recipe for target 'CMakeFiles/_BOPAlgo.dir/Unix/x86_64-Release-/BOPAlgoPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_BOPAlgo.dir/Unix/x86_64-Release-/BOPAlgoPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:9471: recipe for target 'CMakeFiles/_BOPAlgo.dir/all' failed
make[1]: *** [CMakeFiles/_BOPAlgo.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

```

tpaviot, 2020-07-14:
The current pythonocc master branch targets occt-7.4.0, and successfully compile on OSX/Linux/Windows according to Azure pipeline. What's your platform ? Are you sure you pass the correct occt path to cmake ?

hoba87, 2020-07-15:
Ok, sorry, the try meant to be with oct-7.4.0 was still oct-7.4.0p1, with oct-7.4.0 compilation is working

CyrilWaechter, 2020-08-31:
Hi,
I got same issue. opencascade 7.4.0p1 is now default on Arch. Is there any schedule/plan to update to 7.4.0p1 ? If I can help in some way let me know but my knowledge of c++, opencascade and pythonocc-core is little. I use it mainly from FreeCAD and Blender BIM Addon.

tpaviot, 2020-08-31:
@CyrilWaechter AFAIK 7.4.0 is the latest opencascade public release (https://www.opencascade.com/content/latest-release), released on october 2019. 7.4.0p1 was tagged 5 months ago on the occt git repository, but there's not any release. Where did the Arch team get 7.4.0p1 source code (to access to git repos you need to register an account and sign a CLA, what I always refused to do) ?

tpaviot, 2020-08-31:
ok, 7.4.0p1 is downloaded here https://github.com/archlinux/svntogit-community/blob/packages/opencascade/trunk/PKGBUILD#L20

CyrilWaechter, 2020-08-31:
I didn't know. As I saw a talk about 7.4.0p1 some time ago on FreeCAD forum I thought it was the latest. As you noticed they download it from here : https://git.dev.opencascade.org/gitweb/?p=occt.git;a=snapshot;h=refs/tags/V7_4_0p1;sf=tgz
https://git.dev.opencascade.org/gitweb/ is apparently accessible without signing any CLA and you find all tags on [summary page](https://git.dev.opencascade.org/gitweb/?p=occt.git;a=summary)

tpaviot, 2020-08-31:
@CyrilWaechter As a temporary workaround, you can skip the ```TreatCompound``` wrapper:
* open the src/SWIG_files/wrapper/BOPALgo.i interface file (https://github.com/tpaviot/pythonocc-core/blob/master/src/SWIG_files/wrapper/BOPAlgo.i);

* remove lines 965 to 978

* save, and compile

Trophime, 2020-09-15:
With the following patch, pythonocc-core smoothly compiles on Debian/Testing with opencascade 7.4.1 debain package.
```
Index: pythonocc-core-7.4.0/src/SWIG_files/wrapper/BOPAlgo.i
===================================================================
--- pythonocc-core-7.4.0.orig/src/SWIG_files/wrapper/BOPAlgo.i
+++ pythonocc-core-7.4.0/src/SWIG_files/wrapper/BOPAlgo.i
@@ -632,17 +632,6 @@ class BOPAlgo_Tools {
 	:rtype: void") PerformCommonBlocks;
 		static void PerformCommonBlocks (const BOPDS_IndexedDataMapOfPaveBlockListOfInteger & theMBlocks,const opencascade::handle<NCollection_BaseAllocator> & theAllocator,BOPDS_PDS & pDS,const opencascade::handle<IntTools_Context> & theContext = opencascade::handle<IntTools_Context>());
 
-		/****************** TreatCompound ******************/
-		%feature("compactdefaultargs") TreatCompound;
-		%feature("autodoc", "* Collect in the output list recursively all non-compound subshapes of the first level of the given shape theS. If a shape presents in the map theMFence it is skipped. All shapes put in the output are also added into theMFence.
-	:param theS:
-	:type theS: TopoDS_Shape
-	:param theMFence:
-	:type theMFence: TopTools_MapOfShape
-	:param theLS:
-	:type theLS: TopTools_ListOfShape
-	:rtype: void") TreatCompound;
-		static void TreatCompound (const TopoDS_Shape & theS,TopTools_MapOfShape & theMFence,TopTools_ListOfShape & theLS);
 
 		/****************** WiresToFaces ******************/
 		%feature("compactdefaultargs") WiresToFaces;
Index: pythonocc-core-7.4.0/src/SWIG_files/wrapper/BOPTools.i
===================================================================
--- pythonocc-core-7.4.0.orig/src/SWIG_files/wrapper/BOPTools.i
+++ pythonocc-core-7.4.0/src/SWIG_files/wrapper/BOPTools.i
@@ -776,6 +776,18 @@ class BOPTools_AlgoTools {
 	:rtype: int") Sense;
 		static Standard_Integer Sense (const TopoDS_Face & theF1,const TopoDS_Face & theF2,const opencascade::handle<IntTools_Context> & theContext);
 
+		/****************** TreatCompound ******************/
+		%feature("compactdefaultargs") TreatCompound;
+		%feature("autodoc", "* Collect in the output list recursively all non-compound subshapes of the first level of the given shape theS. If a shape presents in the map theMFence it is skipped. All shapes put in the output are also added into theMFence.
+	:param theS:
+	:type theS: TopoDS_Shape
+	:param theLS:
+	:type theLS: TopTools_ListOfShape
+	:param theMFence:
+	:type theMFence: TopTools_MapOfShape
+	:rtype: void") TreatCompound;
+		static void TreatCompound (const TopoDS_Shape & theS,TopTools_ListOfShape & theLS, TopTools_MapOfShape * theMFence);
+
 		/****************** UpdateVertex ******************/
 		%feature("compactdefaultargs") UpdateVertex;
 		%feature("autodoc", "* @name Updating the vertex Update the tolerance value for vertex <aV> taking into account the fact that <aV> lays on the curve <aIC>
```

PR #901 (2020-10-13): Removed BOpAlgo_Tools.TreatCompound for compliancy with occt7.4.0p1
Fix #870
closed 2020-10-13 by tpaviot