tpaviot/pythonocc-core #871: Cannot read .stl, .stp from Amazon S3 Bucket

imgeaslikok, 2020-07-16:
I'm trying to read .stl or .stp file from Amazon S3 Bucket by using `from OCC.Extend.DataExchange import read_stl_file, read_step_file` but I cannot. I have passed the file path such as below:
```
s3 = boto3.resource('s3', region_name='eu-central-1')
        bucket = s3.Bucket('bucket')
        obj = bucket.Object(uploadedVolume)
        tmp = tempfile.NamedTemporaryFile()
        
        with open(tmp.name, 'wb') as f:
            obj.download_fileobj(f)
            shape = read_stl_file(tmp.name)
```

But the shape variable is null. How can I read from Amazon S3 bucket?

tpaviot, 2020-07-16:
did you search/find any related post at stackoverflow ?

imgeaslikok, 2020-07-16:
no I can't find. also I have checked this repo's all the issues (opened or closed) and I can't find anything. so I have asked this question in stackoverflow: https://stackoverflow.com/questions/62935633/valueerror-invalid-null-reference-while-using-occ-package

alitokur, 2020-07-26:
when I try to get .stp image from s3 bucket the following error occurs:
`invalid null reference in method 'brepgprop_VolumeProperties', argument 1 of type 'TopoDS_Shape const &'`

closed 2020-07-27 by imgeaslikok