tpaviot/pythonocc-core #872: How can I install the pythonOCC library

b21604935, 2020-07-16:
I need to read the step file in python, but I failed to install the pythonOCC library. I followed the steps in this link ( https://github.com/vinoth136/pythonocc-core ) but it didn't. What do I need to do for the installation. Do you have any articles about the installation? I use the operating system Windows 10 and pycharm.

b21604935, 2020-07-20:
I did all the steps, pycharm gives an error like this. does not detect occ

Collecting package metadata (current_repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - occ

Current channels:

  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

tpaviot, 2020-07-20:
I guess you have to add the dlr-sc channel to the pycharm channels list. Did you read https://docs.anaconda.com/anaconda/user-guide/tasks/pycharm/ ?

b21604935, 2020-07-21:
@tpaviot Thanks teacher.
pycharm needs to be done
"To add a repository, select Manage Repositories, select +, and enter “conda-forge”. Select OK."

closed 2020-07-21 by b21604935