tpaviot/pythonocc-core #876: SetTransparency() expects an extra argument

cristobaltapia, 2020-08-07:
Hi, I was running an older code with the newer version of python-occ (7.4.0) and encounter some problems. One of them is that the function `SetTransparency()` now requires an extra parameter. Before I would only supply two: shape and transparency (value between 0 and 1). Now python-occ gives the following error:

```
display.Context.SetTransparency(board, 0.3)                                                         
TypeError: AIS_InteractiveContext_SetTransparency expected 4 arguments, got 3
```

I looked into the documentation, but could not find any hint regarding the extra parameter. Is this a bug, or is the documentation outdated? In the latter case: which is the third parameter needed?
Thanks!

greeshmaunni1990, 2020-10-12:
@cristobaltapia Hi i am facing the same issue.Did you come up with a solution?? Please reply.

tpaviot, 2020-10-12:
The third agrument is a boolean, set it to ```True```:

```display.Context.SetTransparency(board, 0.3, True)```

closed 2020-10-12 by tpaviot

greeshmaunni1990, 2020-10-13:
I have tried the above solution and i am again getting another error
TypeError: in method 'AIS_InteractiveContext_SetTransparency', argument 2 of type 'opencascade::handle< AIS_InteractiveObject > const &'