tpaviot/pythonocc-core #877: [Question] Is it possible to receive pythonocc-core objects inside my own SWIG wrapper?

Dogrtt, 2020-08-14:
I write small `C++` lib that use `OpenCASCADE`, and I wanna provide possibility to accept and return `pythonocc-core` objects inside my own `SWIG` wrapper. For now if I got some method in `C++` class that return `TopoDS_Shape *`, in `Python` I receive `<Swig Object of type 'TopoDS_Shape *' at 0x00000204F9E32060>`, but I wanna work with `pythonocc-core` classes. Is it possible to achieve that? Should I somehow describe `%typemap` of returned pointer?

rainman110, 2020-08-14:
Yes, it's definitely possible. We do the same for our c++ library tigl and achieve full interoperability with pythonocc.

Look at our implementation here :
https://github.com/DLR-SC/tigl/tree/master/bindings/python_internal

rainman110, 2020-08-14:
The take home message is, that you have to import the TopoDS.i file from pythonocc-core in your Swig interface file. This makes TopoDS_Shape objects known to the swig type system

Dogrtt, 2020-08-14:
@rainman110 Thank you for advise. I take a look into `tigl` bindings, but some things is not clear for me. I didn't find where you `%include <TopoDS.i>`  and how are you find `PythonOCC` libs by `find_package(PythonOCC)`? Anyway, thank you, it looks like I should spend a little bit more time to get the point.

rainman110, 2020-08-14:
Strange, you are right. It was defeinitely required once. This was the first version where it is still included:
https://github.com/DLR-SC/tigl/blob/8116ca324b87e1d35eaf60582eb51a78beb9912b/bindings/python_internal/geometry.i

Maybe use this as an example.

> and how are you find PythonOCC libs by find_package(PythonOCC)?

I wrote a cmake find script:
https://github.com/DLR-SC/tigl/blob/master/cmake/FindPythonOCC.cmake

Dogrtt, 2020-08-14:
@rainman110 Yeah, thank you, I'm figured out how it should be included, but now I got a lot of errors on `FooParserPYTHON_wrap.cxx` building step. Almost all of them is about syntax errors and undeclared identifiers.

```
  [6/10] Building CXX object wrapper\CMakeFiles\_foo_parser.dir\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx.obj
  FAILED: wrapper/CMakeFiles/_foo_parser.dir/CMakeFiles/_foo_parser.dir/FooParserPYTHON_wrap.cxx.obj 
  C:\PROGRA~2\MICROS~1\2019\COMMUN~1\VC\Tools\MSVC\1427~1.291\bin\Hostx64\x64\cl.exe  /nologo /TP -D_foo_parser_EXPORTS -IF:\Programs\OCCT\opencascade-7.4.0\inc -I..\..\..\wrapper -IF:\Programs\Python38\include -IG:\pythonocc-core-master\src\SWIG_files\wrapper /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /Zi /O2 /Ob1 /DNDEBUG /showIncludes /Fowrapper\CMakeFiles\_foo_parser.dir\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx.obj /Fdwrapper\CMakeFiles\_foo_parser.dir\ /FS -c wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3339): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3340): error C2143: syntax error: missing ';' before 'delete'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3344): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3385): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3386): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3390): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3433): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3434): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3438): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3481): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3482): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3486): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3524): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3525): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3529): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3572): error C3861: 'OCC_CATCH_SIGNALS': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3577): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3624): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3625): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3629): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3658): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3659): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3663): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3700): error C2065: 'OCC_CATCH_SIGNALS': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3701): error C2146: syntax error: missing ';' before identifier 'result'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3705): error C3861: 'process_exception': identifier not found
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3749): error C2065: 'TColStd_ListOfInteger': undeclared identifier
[... cut ...]
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3902): error C2065: 'Message_Report': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3902): error C2059: syntax error: ')'
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3905): error C2065: 'TColStd_HSequenceOfExtendedString': undeclared identifier
F:\CPP\FooParser\out\build\x64-Release\wrapper\CMakeFiles\_foo_parser.dir\FooParserPYTHON_wrap.cxx(3905): fatal error C1003: error count exceeds 100; stopping compilation
```

rainman110, 2020-08-14:
Your just missing the includes. You have to include some OCCT header files via

```
%{
#include<Standard_ErrorHandler.hxx>
#include<Standard_Failure.hxx>
%}
```

You proably need some more than these.

Dogrtt, 2020-08-17:
@rainman110 Thank you. Actually I included a lot of OpenCASCADE headers inside my `*.i` file to make it work, such as `Message_Alert.hxx`, `TColStd_HArray1OfByte.hxx`, etc. Now the main problem is type mapping of `std::vector<TopoDS_Shape *> &myInputArg` arguments to `Python` types. I frustrated by `error C3861: 'process_exception': identifier not found`, but I think it's more about `SWIG` than `pythonocc-core`.