tpaviot/pythonocc-core #879: ModuleNotFoundError: No module named 'OCC.gp'

jayrajsolanki07, 2020-08-18:
hello,
i am using another environment (not root) and i am facing  " ModuleNotFoundError: No module named 'OCC.gp' " this error;
 initially , i was facing" No module named 'OCC'" but which i was able to solve..
can anyone help me?? what is these errors ? and how to solve this ????

Thanks in advance ..

Maozerhouni, 2020-08-27:
Hi,
I had the same issue. My workaround was to copy the content of my Core Folder in the OCC Folder.

mohamed0123, 2020-09-02:
I think it's moved to Core package use it as below
    `from OCC.Core.gp import `

tpaviot, 2020-09-05:
@mohamed0123 is right

closed 2020-09-05 by tpaviot

Ramisay, 2021-05-23:
@jayrajsolanki07  , Please , how did you solve the problem of : " No module named 'OCC'" ?  
Thank you in advance.

thomasTUK, 2021-05-29:
Also wondering. I dont get it to work...

tpaviot, 2021-05-31:
Have a look at the pythonocc-demos to see how to import occ modules https://github.com/tpaviot/pythonocc-demos/tree/master/examples

Basic occt modules can be loaded using
```python
from OCC.Core.XXX import YYY
```
Don't forget the ``.Core`` subpackage.

ts0ron, 2021-07-28:
Hey @tpaviot ,

I looked at number of issues about this question and yet the change from OCC. ... to OCC.Core does not change the 'no module found' error for the 'OCC' prefix. 
Installed pythonocc-core using the next command:

conda install -c dlr-sc pythonocc-core=7.4.0

Would appreciate any assistence,
Thanks.