tpaviot/pythonocc-core #881: IfcOpenShell compatibility

Maozerhouni, 2020-08-27:
Hi,
I use IfcOpenShell to pare Ifc files. Trying the ifcopenshell-academy examples, methods have changed in the new pythoncore-occ. What is the new way to display shapes from ifcfile in OCC ?

Thanks

tpaviot, 2020-09-05:
See https://github.com/tpaviot/pythonocc-demos/blob/web3d2020/jupyter_notebooks/ifc_display_basic_file.ipynb

closed 2020-09-05 by tpaviot

AlonSpinner, 2021-11-09:
Hi, 

I'm a little new to the software world.
I was wondering what is the best way to "keep updated" or "keep stable" and perhaps also contribute? 

Running Linux, here is what I did. It 'mostly works':

$ conda create --name=pyoccenv python=3.7
$ conda install -c conda-forge pythonocc-core=7.4.1
$ conda install -c conda-forge -c oce -c dlr-sc -c ifcopenshell ifcopenshell #installed ifcopenshell 0.6
$ conda install -c conda-forge pythreejs
$ conda install -c conda-forge matplotlib #easy way to get pyqt5
$ conda install -c conda-forge notebook
$ pip install -U ipykernel #allows debugging in vs code
$ pip install git+https://github.com/tpaviot/pythonocc-utils

Thanks