tpaviot/pythonocc-core #886: Fails to run testsuite for latest release

Trophime, 2020-09-15:
Trying to run the testsuite on Debian/Testing with opencascade 7.4.1 from debian package, I've end up with the following error:
```
ERROR: test_read_step_file_multiple_shape_as_root (core_extend_dataexchange_unittest.TestExtendDataExchange)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/build/pythonocc-core-7.4.0/test/core_extend_dataexchange_unittest.py", line 60, in test_read_step_file_multiple_shape_as_root
    t = read_step_file(STEP_MULTIPLE_ROOT, as_compound=True)
  File "/build/pythonocc-core-7.4.0/debian/tmp/usr/lib/python3/dist-packages/OCC/Extend/DataExchange.py", line 61, in read_step_file
    raise FileNotFoundError("%s not found." % filename)
FileNotFoundError: ./test_io/stp_mulitple_shp_at_root.stp not found.

----------------------------------------------------------------------
```

Dunno what to do?
Is the file really missing?

tpaviot, 2020-12-01:
The file is not missing from the repository.

tpaviot, 2021-04-15:
Fixed for a long time.

closed 2021-04-15 by tpaviot