tpaviot/pythonocc-core #892: occt package "OpenGl" seems to be missing from python-occ

cafhach, 2020-09-28:
I would like to use OpenGl_ShaderManager to build a custom shader in a python-occ app. But there doesn't seem to be a Python wrapper for it. Did it slip through or was this done on purpose?

tpaviot, 2020-10-12:
Indeed the OpenGL package is not wrapped. I'm about to commit a fix

tpaviot, 2020-10-12:
@cafhach Did you try to use the ```Graphic3d_ShaderProgram``` class, which is actually wrapped ? Are you sure you need the OpenGl base class ```OpenGl_ShaderManager``` ?

tpaviot, 2020-10-12:
There's an example in the pythonocc demo suite https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_visualization_glsl.py It show how to deal with shaders, what else would you need ?