tpaviot/pythonocc-core #898: DynamicType() no longer wrapped on Geom_ classes.

aothms, 2020-10-07:
This used to work fine prior to 7.4

```
surface = OCC.BRep.BRep_Tool.Surface(face)
surface_type = surface.GetObject().DynamicType().GetObject().Name()
if surface_type != "Geom_Plane":
    ...
```

In 7.4 DynamicType() returns an opaque SWIG object:

```
>>> surface.DynamicType()
<Swig Object of type 'opencascade::handle< Standard_Type > *' at 0x04903DB8>
>>>
>>> dir(surface.DynamicType())
['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__int__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'acquire', 'append', 'disown', 'next', 'own']
```

I did notice that `DownCast()` returns `None` on "invalid" casts.

```
>>> Geom.Geom_Plane.DownCast(surface)
<class 'Geom_Plane'>
>>> repr(Geom.Geom_CylindricalSurface.DownCast(surface))
'None'
```

So there is a way to do type checking, but I tend to rely in my code quite a bit on the DynamicType() to be properly wrapped.

tpaviot, 2020-10-10:
The ```Standard_Type``` class is explicitly excluded from the wrapper, for years I guess. Let me try to expose it and check if there still are issues.

tpaviot, 2020-10-12:
Fixed by commits bb68a00  and c1e9d49

closed 2020-10-12 by tpaviot