tpaviot/pythonocc-core #899: Display3d.GetImageData inputs

cylopa, 2020-10-12:
Could we pass the width and height to the GetImageData method? Currently it only takes buffer type. 

Current behaviour:
display3d.SetSize(width, height)  #This is an expensive operation as it creates another window.
image_data = display3d.GetImageData(buffer_type)

Desired behaviour:
image_data = display3d.GetImageData(width, height, buffer_type)  # If we could pass with and height to the .ToPixMap instead of the windows current size we would not need to re-create the window beforehand.

tpaviot, 2020-10-12:
Yes, this could be done, the ```GetImageData``` python wrapper is defined here https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Visualization.i#L79 I guess it's possible to add the width and height parameters.

tpaviot, 2020-10-12:
@cylopa could you please provide a full example including the desired change so that I can test a fix

tpaviot, 2020-10-12:
Indeed the call to the ```SetSize``` method might be mandatory, I'd like to check that

cylopa, 2020-10-12:
Incoming example in...

cylopa, 2020-10-12:
<details><summary>Current behaviour example</summary>
<p>

```python
import sys
import time

from PySide2.QtWidgets import QApplication, QMainWindow, QWidget
from PySide2.QtGui import QPainter, QImage
from PySide2.QtCore import QPoint

from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.Graphic3d import Graphic3d_BT_RGB
from OCC.Core.AIS import AIS_Shaded, AIS_Shape
from OCC.Display.OCCViewer import Viewer3d
from OCC.Core.gp import gp_Pnt

class ImageWidget(QWidget):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)

        self.origin_point = QPoint(0, 0)

        self.python_occ = Viewer3d(None)
        self.python_occ.Create()
        self.python_occ.display_triedron()

        ais_cube = AIS_Shape(BRepPrimAPI_MakeBox(gp_Pnt(0.0, 0.0, 0.0), 1.0, 1.0, 1.0).Shape())
        self.python_occ.Context.Display(ais_cube, AIS_Shaded, -1, False)
        self.python_occ.FitAll()

    def paintEvent(self, event):
        q_painter = QPainter(self)

        q_device = q_painter.device()
        width = q_device.width()
        height = q_device.height()

        t0 = time.perf_counter()

        self.python_occ.SetSize(width, height)

        t1 = time.perf_counter()

        image_data = self.python_occ.GetImageData(Graphic3d_BT_RGB)

        t2 = time.perf_counter()

        q_image = QImage(image_data, width, height, width * 3, QImage.Format_RGB888)

        # GetImageData returns an upside down image. So we flip it back.
        # I guess it has something to do with Image_PixMap.SetTopDown

        q_painter.translate(0, q_device.height())
        q_painter.scale(1.0, -1.0)

        t3 = time.perf_counter()

        q_painter.drawImage(self.origin_point, q_image)

        t4 = time.perf_counter()

        q_painter.end()

        print("SetSize      :{:.6f}".format(t1 - t0))
        print("GetImageData :{:.6f}".format(t2 - t1))
        print("QImage       :{:.6f}".format(t3 - t2))
        print("drawImage    :{:.6f}".format(t4 - t3))
        print()

class MainWindow(QMainWindow):
    def __init__(self, width, height, parent=None):
        self.q_app = QApplication.instance()
        if self.q_app is None:
            self.q_app = QApplication(sys.argv)
        super().__init__(parent=parent)

        self.setWindowTitle("PythonOCC - OffScreen Example")
        self.resize(width, height)

        self.image_widget = ImageWidget()
        self.setCentralWidget(self.image_widget)

    def event_loop(self):
        self.show()
        sys.exit(self.q_app.exec_())

if __name__ == "__main__":
    window = MainWindow(1024, 768)
    window.event_loop()

```

</p>
</details>

<details><summary>Desired behaviour example</summary>
<p>

```python
import sys
import time

from PySide2.QtWidgets import QApplication, QMainWindow, QWidget
from PySide2.QtGui import QPainter, QImage
from PySide2.QtCore import QPoint

from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.Graphic3d import Graphic3d_BT_RGB
from OCC.Core.AIS import AIS_Shaded, AIS_Shape
from OCC.Display.OCCViewer import Viewer3d
from OCC.Core.gp import gp_Pnt

class ImageWidget(QWidget):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)

        self.origin_point = QPoint(0, 0)

        self.python_occ = Viewer3d(None)
        self.python_occ.Create()
        self.python_occ.display_triedron()

        ais_cube = AIS_Shape(BRepPrimAPI_MakeBox(gp_Pnt(0.0, 0.0, 0.0), 1.0, 1.0, 1.0).Shape())
        self.python_occ.Context.Display(ais_cube, AIS_Shaded, -1, False)
        self.python_occ.FitAll()

    def paintEvent(self, event):
        q_painter = QPainter(self)

        q_device = q_painter.device()
        width = q_device.width()
        height = q_device.height()

        t0 = time.perf_counter()

        image_data = self.python_occ.GetImageData(width, height, Graphic3d_BT_RGB)

        t1 = time.perf_counter()

        q_image = QImage(image_data, width, height, width * 3, QImage.Format_RGB888)

        t2 = time.perf_counter()

        q_painter.drawImage(self.origin_point, q_image)

        t3 = time.perf_counter()

        q_painter.end()

        print("GetImageData :{:.6f}".format(t1 - t0))
        print("QImage       :{:.6f}".format(t2 - t1))
        print("drawImage    :{:.6f}".format(t3 - t2))
        print()

class MainWindow(QMainWindow):
    def __init__(self, width, height, parent=None):
        self.q_app = QApplication.instance()
        if self.q_app is None:
            self.q_app = QApplication(sys.argv)
        super().__init__(parent=parent)

        self.setWindowTitle("PythonOCC - OffScreen Example")
        self.resize(width, height)

        self.image_widget = ImageWidget()
        self.setCentralWidget(self.image_widget)

    def event_loop(self):
        self.show()
        sys.exit(self.q_app.exec_())

if __name__ == "__main__":
    window = MainWindow(1024, 768)
    window.event_loop()

```

</p>
</details>

tpaviot, 2020-10-13:
accordingly updated the core_offscreen_rendere example at https://github.com/tpaviot/pythonocc-demos/commit/4832a91404272f2da27bb2d454718e78e9512a35

cylopa, 2020-10-13:
Thank you @tpaviot. I had been using conda packages so I will build and try it.

tpaviot, 2020-10-13:
closing the issue, please reopen if something is wrong with the fix

closed 2020-10-13 by tpaviot