tpaviot/pythonocc-core #904: Reading PMI and GD&T information from STEP files

theveloped, 2020-10-18:
I'm wondering if it is possible to use `STEPCAFControl_Reader` to extract the dimension and tolerance information (e.g. `XCAFDoc_Dimension`) that can be embedded in a STEP file. In the examples I've found various examples that cover extracting colors but not any other information.

tpaviot, 2020-12-01:
I guess it's possible, but never did it. If you manage to get something, please post yor feedback here

varunkaarthik, 2023-04-20:
Hey @theveloped,

Did you manage to find a solution for this?

onkarp12, 2024-05-08:
Hey @theveloped,
Did you manage to find a solution for this?

tpaviot, 2024-05-09:
@onkarp12 can you please provide a test step file including PMI and GDT information for testing

onkarp12, 2024-05-09:
[StepFilesWithPMI.zip](https://github.com/tpaviot/pythonocc-core/files/15257974/StepFilesWithPMI.zip)
@tpaviot  Please find the files. Thanks for looking into this. Appreciate it!

Regards,
Onkar

tpaviot, 2024-05-09:
> [StepFilesWithPMI.zip](https://github.com/tpaviot/pythonocc-core/files/15257974/StepFilesWithPMI.zip) @tpaviot Please find the files. Thanks for looking into this. Appreciate it!
> 
> Regards, Onkar

:+1:

lukasamsa, 2024-05-23:
I spent quite some time researching this topic and trying to find a solution on how to extract the graphic presentation of the GDT. Using the OCC it is pretty easy to extract the shapes of the REFERENCE shapes, which means we are left with the faces on which the GDT is appended. On the other hand it is much harder, if even possible to somehow extract the exact geometry of the GDT dimension itself.

For example:
`gdt_tool = XCAFDoc_DocumentTool.DimTolTool(doc.Main())
dim_tol_labels = TDF_LabelSequence()
gdt_tool.GetDimensionLabels(dim_tol_labels)
# gdt_tool.GetDatumLabels(labels) -> returns the Datum labels
# gdt_tool.GetGeomToleranceLabels(labels) -> returns the Geometric Tolerance labels`

extracts the labels of the dimensions, but when I tried to extract the shape of the GDT, it always returns just the reference shape. 

The .GetPresentation() function returned a TopoDS_Compound, which seems to be empty: no faces, no edges, no nothing... 

Another possible way could be this one:
[image]

which I think isn't yet feasible with the pythonocc, since the NCollection_IndexedDataMap() is not wrapped... 

Similar issue was opened and discussed in this repo: [(https://github.com/usnistgov/STP2X3D/issues/7)]. 

Although this comment isn't anything new and didn't solve much, I wanted to open a discussion, which might spark an idea for someone... I think that presentation of the GDT will be an important factor, since Model Based Definitions are becoming a standard practice these days.

tpaviot, 2024-05-24:
This place https://github.com/Open-Cascade-SAS/OCCT/blob/master/src/XDEDRAW/XDEDRAW_GDTs.cxx is certainly a good place to start with GDT. A port of these functions to python would be great

lukasamsa, 2024-05-24:
I've seen a couple of examples using this method: 

c++:
```
Handle(TCollection_HAsciiString) aName = new TCollection_HAsciiString(argv[4]);
// Dimension
Handle(XCAFDoc_Dimension) aDimension;
if (aLabel.FindAttribute(XCAFDoc_Dimension::GetID(), aDimension))
{
  Handle(XCAFDimTolObjects_DimensionObject) anObj = aDimension->GetObject();
  anObj->SetPresentation(aPresentation, aName);
  aDimension->SetObject(anObj);
}
```
which can be easily translated to python, but I stumbled upon an issue. The `.FindAttribute(XCAFDoc_Dimension::GetID(), aDimension)` expects a handle as a second parameter. As far as I know, you can't really create a handle in python as the function expects us to, so if anyone knows any workarounds it would be great.