tpaviot/pythonocc-core #905: trying to find the normal for a TopoDS_Edge using the class BRepLProp_CLProps()

ANkhan-93, 2020-10-19:
I have a "cylindrical face" which intersects with a "conical face" at an angle resulting in a TopoDS_Edge; as can be seen in the image below. The resulting edge is colored white. For this operation I used the class "**BRepAlgoAPI_Section()**".

[image]

After that I successfully used the GCPnts_UniformAbscissa() to find out equally-spaced points on this edge, as shown in figure below:

[image]

**Now I want to find the normal vector at each of these points, but I m getting an error and I use the following code.**
 
hatch_curve = BRepAdaptor_Curve(TopoDS_Edge)
properties = BRepLProp_CLProps( hatch_curve, BRepLProp_CurveTool.FirstParameter(hatch_curve), 3, 0.005)
_I am a bit confused about the the last parameter 'Resolution', but it works as 0.005_

normal = gp_Dir()
properties.Normal(normal)

**When I run this, I get the following error:**

LProp_CLProps::Normal(...):Curvature is null or infinity
wrapper details:
  * symname: BRepLProp_CLProps_Normal
  * wrapname: _wrap_BRepLProp_CLProps_Normal
  * fulldecl: void BRepLProp_CLProps::Normal(gp_Dir &)

**Therefore I checked the curvature using the method:**
properties.Curvature()

and the curvature is 0.0
So I am confused as to how should I find the normal to this curve?

tpaviot, 2020-12-01:
Can you please provide a full script that I can run on my machine. The issue may either come from a misuse of the library or a bug into occt itself, which is possibly fixed in the latest release.

ANkhan-93, 2020-12-01:
Unfortunately I do not have the script available to me anymore, as I deleted it and used another module within the OCC to find the normal vectors to the surface at the points along the white edge.

Thanks

closed 2020-12-02 by tpaviot

qigong777, 2022-08-22:
> Unfortunately I do not have the script available to me anymore, as I deleted it and used another module within the OCC to find the normal vectors to the surface at the points along the white edge.
> 
> Thanks

Hi, could you tell me what module you use?