tpaviot/pythonocc-core #907: STEP export with informations

BrayanHrz, 2020-10-27:
I'm working on the export of models to step.

I exported solid models to step but there is some information that has been lost e.g. when I export a cone with ribs, there is no entity corresponding to a boolean operation, or a boolean result. However, according to www.steptools.com/stds/stp_aim/html/ this kind of entity actually exists in the AP used in my script.

So I tried to create a document in order to keep all the operations processed but it seems impossible.
My last try was using the StepShape module with its boolean operators but I just cannot export the obtained solids into a step file.

Can anyone help me to fix one of the issues ?

BrayanHrz, 2020-10-27:
[image]
[image]
Here attached the list of entity used to save my file to step format, my solid and the kind of entity I would like to keep.
[image]

waterbug, 2020-10-27:
Even though the boolean_result entity definitely exists in STEP schemas -- it is defined in the STEP geometry resource, part 42 (ISO 10303-42), so it appears in all the application protocols that have geometry (ap203, ap214, ap242, etc.) -- that doesn't necessarily mean it is supported by STEP translators used by actual CAD systems.  In fact, since the boolean_result entity does not appear in any of the STEP files used by the CAX-IF (CAx Implementers' Forum), where STEP translators are tested for interoperability, there is no evidence that it has been used or tested in actual code.  Further, in the rather extensive "recommended practices" documents published by the CAX-IF, there is no mention whatsoever of how to implement the boolean_result entity.  Granted, it does appear in the StepShape wrapping in pythonocc, but even there it may not actually be "implemented", it might just have been generated automatically from the schema.  Looking at the boolean_result entity, its first_operand and second_operand attributes reference the shapes to which the boolean operator was applied -- in other words, you would have to have the full geometric definition of the solids that were operated on also defined in your STEP file.  In googling for STEP file examples, I did not find anything but an example from the "IFC" world (which uses the STEP Express and part 21 technology, but uses simplified versions of STEP schemas) of an "IFCBOOLEANRESULT", which is obviously copied from the STEP boolean_result entity, used in an example file called "bath-csg-solid.ifc".  So it appears to be supported in the IFC world but it may be that constructive solid geometry is more important there than in the general CAD world.

BrayanHrz, 2020-10-28:
> Even though the boolean_result entity definitely exists in STEP schemas -- it is defined in the STEP geometry resource, part 42 (ISO 10303-42), so it appears in all the application protocols that [...]

Thank you for your complete and quick reply,
I will check CAX-IF, it seems to be a complete description of interoperability.
boolean_result was just an example, indeed after have explored OCCT and pythonOCC library, circular pattern seems not to be implemented, (if it is, it is not named like this). My goal is to provide high level STEP files in order to keep the information of computed operations, to finally get the solid model.

Many thanks again,
Brayan

closed 2020-12-01 by tpaviot