tpaviot/pythonocc-core #909: Export annotated shapes to step

BrayanHrz, 2020-10-29:
I am trying to export annotated models into step file. But the problem is the following :
I am using XCAFDoc and I added a note, I tested it with .IsAnnotated()  and when I import it into FreeCAD, there is no note, moreover, even in the step file there is no comment. I tried to export it following various APs but the result is still the same.
Can anyone help me to make the annotations exported ?

tpaviot, 2020-12-01:
Can you please provide some python code I could run to reproduce the issue.

BrayanHrz, 2020-12-02:
`# Test data.
dx = 5.
dy = 5.
dz = 10.
radius = 10.

# Function that returns a color based on rgb system, 0 <= r, g, b <= 1.
def rgbcolor(r, g, b):
    return Quantity_Color(r, g, b, Quantity_TOC_RGB)
red = rgbcolor(1, 0, 0)
green = rgbcolor(0, 1, 0)
blue = rgbcolor(0, 0, 1)
yellow = rgbcolor(1, 1, 0)
purple = rgbcolor(1, 0, 1)

# Create an handle to a document.
doc = TDocStd_Document(TCollection_ExtendedString("color-doc"))
shapeTool = XCAFDoc_DocumentTool.ShapeTool(doc.Main())
colors = XCAFDoc_DocumentTool.ColorTool(doc.Main())
notes = XCAFDoc_DocumentTool.NotesTool(doc.Main())

# Test shapes.
sphere = BRepPrimAPI_MakeSphere(gp_Pnt(10, 10, 0), radius).Shape()
box = BRepPrimAPI_MakeBox(dx, dy, dz).Shape()
common = BRepAlgoAPI_Common(box, sphere).Shape()

# Create the labels and add the targeted shapes in it.
boxLabel = shapeTool.AddShape(box)
sphereLabel = shapeTool.AddShape(sphere)
# Here, the 'False' argument is necessary to display the color. Boolean results are not considered as pure shapes here but as compounds.
commonLabel = shapeTool.AddShape(common, False)

# Set the color labels.
colors.SetColor(boxLabel, red, XCAFDoc_ColorGen)
colors.SetColor(sphereLabel, yellow, XCAFDoc_ColorGen)
colors.SetColor(commonLabel, purple, XCAFDoc_ColorGen)

# Set noter label.
commonComment = notes.CreateComment(TCollection_ExtendedString('Intes'), TCollection_ExtendedString('29_10'), TCollection_ExtendedString('common is a boolean result'))
comm = notes.AddNote(commonComment.Label(), commonLabel)

WS = XSControl_WorkSession()
stepWriter = STEPCAFControl_Writer()
Interface_Static_SetCVal("write.step.schema", "AP242DIS")
Interface_Static_SetIVal("write.assembly.mode", 0)
stepWriter.Transfer(doc, STEPControl_AsIs)
status = stepWriter.Write("attributes.stp")

if status != IFSelect_RetDone:
    raise AssertionError("load failed")`

BrayanHrz, 2020-12-02:
I'm sorry for the format, I am not familiar with github. 
Here is  a part of my code, you can run it and notice that in the step file, although there is no "ANNOTATED SHAPE" entity, commonLabel.IsAnnotated() returns True.

tpaviot, 2020-12-02:
Right I confirm what you say. I have to check if this comes from a misuse or a limitation in the upstream occt library

tpaviot, 2020-12-02:
Are you sure the ```ANNOTATED SHAPE``` entity is part of the official STEP standard ? which AP ? I can find any occurence of such entiy name in official express schemas (https://www.cax-if.org/cax/cax_express.php)

BrayanHrz, 2020-12-02:
The entity is not annotated shape but annotation_occurrence (see screenshot attached), i'm sorry.
[image]

You can find some example with the following link : [](https://www.cax-if.org/cax/cax_stepLib.php); **IO1-CM-214**.
I did some research on this issue and it seems to be an issue of OCCT library.

BrayanHrz, 2020-12-02:
[https://www.cax-if.org/cax/cax_stepLib.php](url) the link has not been sent.

EmJay276, 2022-03-04:
I'm running into the same issue, is there any solution for adding a Note and exporting it to STEP?