tpaviot/pythonocc-core #910: Summary for curated list of code CAD

Irev-Dev, 2020-10-31:
Hey Thomas 👋 ,

I tweeted at you earlier

I finally got around to putting together a list of all of the code-CAD projects that I know of [here](https://kurthutten.com/blog/curated-code-cad), and also on [Github](https://github.com/Irev-Dev/curated-code-cad).

I've put a short summary of each one, but it's hard to do them all justice. for pythonOCC I've only got:
Python-based, Also uses OpenCascade.

Not the right repo but I also have got the following for openCascade:
It's a c++ library that a number the projects below wrap.

If you would like to give me a summary of the project, and perhaps some points of how it differs from other similar projects would be happy to include it. I think max three paragraphs would be about right. Open a PR on reply to this issue and I'll include it.

tpaviot, 2020-11-09:
ok, please send me a reminder by the end of the week if ever I forget to submit a short text

Irev-Dev, 2020-11-09:
Awesome, will do. Thanks @tpaviot 
A quick extra note is I've changed the OpenCascade description a bit since I opened this issue. Sorry, I forgot to come back and update it. it's now:
It's a c++ library that a number the projects below wrap. OpenCascade uses a Boundary representation (B-rep) kernel, which is a powerful way representing solid geometry, this is a point of difference for some many of the other projects that use a polygon mesh.

Irev-Dev, 2020-11-20:
Hey @tpaviot, Here's a reminder. I forgot too.