tpaviot/pythonocc-core #911: Problem running examples : QWindowsGLContext::makeCurrent: SetPixelFormat()

Jojain, 2020-11-01:
Hello, 
I run in a problem when I run the examples of pythonocc, for example if I run the core_helloworld.py file, the box shows itself but I get the error message ; **QWindowsGLContext::makeCurrent: SetPixelFormat() failed (Le format pixel nÆest pas valide.)**
And if shift+click it crashes, I've seen that someone else had the same problem which hasn't been fully resolved : https://github.com/tpaviot/pythonocc-core/issues/640

For information I run the example on a conda virtual environment with python 3.7, my GUI manager is pyqt5 (I have tried with Pyside2 with the same results) I also tried the same thing with python 3.6, same results.
I run on W10 with a GeForce GTW 1060 6GB, drivers up to date.

I tried the same procedure on a laptop W10 Python 3.7, pyqt5, graphics : Intel (R) UHD Graphics 620 and it runs without any troubles without the error message cited before.
I guess there is something wrong about the graphics but I'm really not familiar with any of this.

If someone has any hints I would be happy to try them out, thank you.

bitbybit3d, 2020-11-19:
I also get this, just like you.  And in my laptop, if use PySide2, will get it, but success if change to PyQt5.

But in my computer, neither pyside2 nor pyqt5 works.

spl1inter, 2020-11-24:
Same issue under MinGW64. Most of  the demos still work though (using PyOCC 18.2 over PyQt5 as PySide2 seems broken on MinGW64). The error appears when anything refreshes the Qt window - like loading a menu widget at the start or resizing the window later.

[image]

The problem is more serious over at ifcopenshell where the same error seemingly prevents anything from being displayed at all, so I'd be interested ideas to fix it.

tpaviot, 2020-12-03:
@Jojain @bitbybit3d @spl1inter can you please report the pythonocc-core version you've been using. Indeed there have been recent fixes to the qt display, be sre you run the latest 7.4.1 version.

spl1inter, 2020-12-10:
Sorry for the delayed response, frankly I was busy on something MinGW-y with IFCOpenShell and then FreeCAD and wasn't paying attention :(

I thought that I posted this before:
> using PyOCC 18.2 over PyQt5..

Let me know if you need anything else.

geraldozzz, 2021-01-27:
I also met the same problem, but only on Windows 7, and it works fine on Windows 10.

The system will crash once paintEvent in qtViewer3d exec, and the exact line caused the problem is:

`painter = QtGui.QPainter(self)`

tpaviot, 2021-04-14:
This should be fixed by commit a028f6063fb1f2506d7ab42802f565c74019d1e7

closed 2021-04-14 by tpaviot