tpaviot/pythonocc-core #912: new release?

greyltc, 2020-11-02:
Any chance you folks could cut a release that will work with the OCCT 7.4.0p1 code base?
https://github.com/tpaviot/pythonocc-core/issues/870

tpaviot, 2020-11-09:
that makes sens, it's coming

adam-urbanczyk, 2020-11-09:
Maybe it is better to go directly to 7.5 ? It should be soon available in conda-forge.

tpaviot, 2020-11-09:
Yes, support for 7.5 is almost ready yet on my laptop. Anyway there have been some work that has been contributed since the last pythonocc release, and I'm not absolutely confident that there won't be any regression when moving up to 7.5. So a 7.4.1 release would be a way to ensure this work made available, and a final point do the 7.x.x series. I'm not notified by the conda-forge channel, please let me know when the occt7.5.0 conda package is available

tpaviot, 2020-11-10:
done https://github.com/tpaviot/pythonocc-core/releases/tag/7.4.1

closed 2020-11-10 by tpaviot

greyltc, 2020-11-10:
Thank you! I'm sure I'll be back soon asking for a 7.5 release :-)
FYI my packaging for Arch and friends is here: https://aur.archlinux.org/packages/python-occ/

adam-urbanczyk, 2020-11-10:
It is already available in conda-forge @tpaviot .

tpaviot, 2020-11-10:
yes @adam-urbanczyk I saw that yesterday, you did an impressive work in such short time congrats !