tpaviot/pythonocc-core #913: XCAFDoc_DimTolTool class does not exist

BrayanHrz, 2020-11-03:
I'm  attributing some data to my solid models, I already attributed materials and colors but when I wanted to attribute geometric data, the error 
AttributeError: 'SwigPyObject' object has no attribute 'SetDimTol'
swig/python detected a memory leak of type 'opencascade::handle< XCAFDoc_DimTolTool > *', no destructor found.
occured. But when I read the open cascade  doc, I find this attribute I'm trying to use. Then I checked the .py file and indeed, I did not find any class called "XCAFDoc_DimTolTool". I'm comparing this structure with e.g. XCAFDoc_ColorTool  that is the same in the occt doc but not in the .py file. Can we attribute geometric data to shapes ?

tpaviot, 2020-11-06:
which pythonocc version do you use ?

tpaviot, 2020-11-09:
I checked that this class is actually wrapped in the current master branch.

closed 2020-11-09 by tpaviot