tpaviot/pythonocc-core #915: conda packages for release 7.4.1

tpaviot, 2020-11-10:
@looooo and @rainman110 Do you plan to update your conda channels with this new release ? It could be great if you can provide conda packages !

rainman110, 2020-11-10:
What opencascade version should we use as a package dependency?

tpaviot, 2020-11-10:
You can keep the same as the previous version 7.4.0 or upgrade to 7.4.0p1

tpaviot, 2020-11-10:
unittests pass for both 7.4.0 and 7.4.0p1
7.4.0p1 is not an official release though, it's only a tag on the occt git master branch, I suggest to keep the same occt ie 7.4.0

looooo, 2020-11-10:
Normally the conda-forge-bot creates a PR for new releases. I think we have to wait a little bit until this happens.

rainman110, 2020-11-10:
The packages have been built for python 3.6 and 3.7 on the DLR-SC channel. Feel free to test them

tpaviot, 2020-11-10:
@looooo perfect, let's wait

@rainman110 yes, I've been monitoring the build process on your travis/appveyor accounts. I wait for everything to complete before testing

looooo, 2020-11-12:
https://github.com/conda-forge/pythonocc-core-feedstock/pull/3
@tpaviot there is an experimental dependencies analysis included. Maybe you can have a look.

mmone, 2020-11-16:
The bot has created the PR in https://github.com/conda-forge/pythonocc-core-feedstock/pulls.

tpaviot, 2020-12-01:
Seem that packages for release 7.4.1 are available on conda-forge, thank you

closed 2020-12-01 by tpaviot