tpaviot/pythonocc-core #916: segfault in "asserts that handling of OCC exceptions is handled correctly caught" test

greyltc, 2020-11-10:
```
Test gp_Vec division by a float number or an integer ... ok
test_handle_standard_transient_copy (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_handling_exceptions (core_wrapper_features_unittest.TestWrapperFeatures)
asserts that handling of OCC exceptions is handled correctly caught ... Segmentation fault      (core dumped) 
```

And the stack trace:
```
Nov 10 17:49:51 REDACTED systemd-coredump[2057111]: Process 2057079 (python) of user 1000 dumped core.
                                                   
                                                   Stack trace of thread 2057079:
                                                   #0  0x00007f812a6251ea _ZNK21BRepBuilderAPI_Sewing8FreeEdgeEi (libTKTopAlgo.so.7 + 0x2091ea)
                                                   #1  0x00007f811e13a195 n/a (/home/REDACTED/git/python-occ/src/pythonocc-core-7.4.1/build/tmp/usr/lib/python3.8/site-packages/OCC/Core/_BRepBuilderAPI.so + 0x6f195)
```

tpaviot, 2020-11-10:
You have to ensure that opencascade is compiled with ```BUILD_RELEASE_DISABLE_EXCEPTIONS=OFF``` otherwise c++ exception are not raised, and cannot be caught by python.

greyltc, 2020-11-10:
Hm. Looks like Arch doesn't build with that flag: https://github.com/archlinux/svntogit-community/blob/packages/opencascade/trunk/PKGBUILD

greyltc, 2020-11-10:
I've raised a request to have that flag added: https://bugs.archlinux.org/task/68585

greyltc, 2020-11-23:
The build flag has been added to Arch's opencascade package, and I'm not seeing this check fail now. Thanks for the tip!

closed 2020-11-23 by greyltc