tpaviot/pythonocc-core #917: port to opencascade-7.5.0

tpaviot, 2020-11-11:
opencascade 7.50 was released a few days ago, conda packages were updated at conda-forge https://anaconda.org/conda-forge/occt

pythonocc wrapper is being updated at https://github.com/tpaviot/pythonocc-core/tree/review/port-to-occt750

The azure build relies on the conda-forge occt 7.5.0 package, it requires some updates (32bit Windows package is not available, and MacOSX build needs some additional tweaks) see https://dev.azure.com/tpaviot/pythonocc-core/_build?definitionId=2&_a=summary&repositoryFilter=2&branchFilter=87%2C87%2C87

tpaviot, 2020-11-11:
@adam-urbanczyk @rainman110 @looooo you might be interested in following this porting effort

tpaviot, 2020-11-11:
According to the latest azure build, it seems all osx/linux/win builds complete, and all unittest pass as well

On my machine, I checked some of the pythonocc-demos, it seems also that everything is ok, I could not find any example that doesn't work with 7.5.0, but I have to check all the scripts to be sure

looooo, 2020-11-11:
A pr for a migration was already created:
https://github.com/conda-forge/conda-forge-pinning-feedstock/pull/935/commits/73b16812ecfc018ece7a6f2ce46dd7c941864a18

But I am not yet sure if it's good to start with the migration now. There are other migrations also ongoing and some stuff needs to be resolved (mainly smesh + vtk9)

tpaviot, 2020-11-11:
@looooo I do not understand what you mean, I can see/download occt7.5.0 from the anaconda conda-forge channel. What do you mean with "migration" ? migration of what ?

looooo, 2020-11-11:
> @looooo I do not understand what you mean, I can see/download occt7.5.0 from the anaconda conda-forge channel. What do you mean with "migration" ? migration of what ?

If the mentioned PR is merged a bot creates pr's for all occt dependent recipes to update to occt 7.5.

tpaviot, 2020-11-11:
ok I got it, thank you

tpaviot, 2021-02-26:
pythonocc-7.4.0 can not be compiled against occt750. I recently tagged 7.5.0rc1 https://github.com/tpaviot/pythonocc-core/releases/tag/7.5.0rc1, this version should be used. Could you @looooo update the conda-forge conda recipe to this version ?

looooo, 2021-02-26:
Should we wait for a release?

looooo, 2021-03-01:
@tpaviot the rc1 tag is failing for me.

tpaviot, 2021-03-02:
@looooo thank you for this feedback, what do you mean exactly with "failing for me" ? Does it successfully compile on your machine ? Does it successfully run all the unittests ? Did you build pythonocc on your own machine or on a compilation farm (conda-forge) ?

looooo, 2021-03-03:
this is the error I see in the PR:
```
CMake Error at CMakeLists.txt:141 (find_package):
  Could not find a configuration file for package "OpenCASCADE" that is
  compatible with requested version "7.5.0".
```

We try to build pythonocc against occt 7.5.1.

looooo, 2021-03-27:
@tpaviot is it possible to update the master branch to work with occt7.5.1?
https://github.com/tpaviot/pythonocc-core/blob/master/CMakeLists.txt#L143

tpaviot, 2021-03-29:
@looooo WIP

tpaviot, 2021-03-30:
@looooo done, merged into master

luzpaz, 2021-03-31:
Can this ticket be closed ?

closed 2021-03-31 by tpaviot