tpaviot/pythonocc-core #920: XCAFDoc_DocumentTool_ShapeTool.AddSubShape returns null

OkanPinar, 2020-11-16:
Hello everyone 

I am trying to create step file with OCC_XCaf_Doc. After creating an empty entry point with XCAFDoc_DocumentTool_ShapeTool.NewShape() or entry point with empty compound with XCAFDoc_DocumentTool_ShapeTool.AddShape(compound, True /* Make it assembly*/), XCAFDoc_DocumentTool_ShapeTool.AddSubShape(assembly_label, shape) return null every time. Example code is following:
 ```
import OCC.Core.STEPCAFControl as OCC_STEP_XCAF_Control
import OCC.Core.STEPControl as OCC_STEP_Control
import OCC.Core.TDocStd as OCC_Doc_STD
import OCC.Core.TCollection as OCC_Type_Collections
import OCC.Core.XCAFDoc as OCC_XCaf_Doc
import OCC.Core.BRepPrimAPI as OCC_Brep_Prim_API
import OCC.Extend.DataExchange as OCC_Extend_Data_Exchange

doc = OCC_Doc_STD.TDocStd_Document(OCC_Type_Collections.TCollection_ExtendedString("pythonocc-doc"))
shape_tool = OCC_XCaf_Doc.XCAFDoc_DocumentTool_ShapeTool(doc.Main())
layer_tool = OCC_XCaf_Doc.XCAFDoc_DocumentTool_LayerTool(doc.Main())

box = OCC_Brep_Prim_API.BRepPrimAPI_MakeBox(20., 40., 60.).Shape()  # Shape to add to document

main_assembly_label = shape_tool.NewShape()
#  main_assembly_label = shape_tool.AddShape(OCC_TopoDS.TopoDS_Compound())
sub_shape_label = shape_tool.AddSubShape(main_assembly_label, box)

step_writer = OCC_STEP_XCAF_Control.STEPCAFControl_Writer()

status = step_writer.Transfer(doc, OCC_STEP_Control.STEPControl_AsIs)

if status != OCC_Extend_Data_Exchange.IFSelect_RetDone:
     raise AssertionError("data transfer failed")
status = step_writer.Write("test.stp")
if status != OCC_Extend_Data_Exchange.IFSelect_RetDone:
    raise Exception('write faild')
```
What is i am missing? I followed the same steps in the following document.

I https://old.opencascade.com/doc/occt-7.4.0/overview/html/occt_user_guides__xde.html#occt_xde_2_2

tpaviot, 2020-11-17:
There's an issue with the way you import classes/functions, try:

```python
from OCC.Core.STEPCAFControl import STEPCAFControl_Writer
from OCC.Core.STEPControl import STEPControl_AsIs
from OCC.Core.TDocStd import TDocStd_Document
from OCC.Core.TCollection import TCollection_ExtendedString
from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool_ShapeTool, XCAFDoc_DocumentTool_LayerTool
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.IFSelect import IFSelect_RetDone

doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))
shape_tool = XCAFDoc_DocumentTool_ShapeTool(doc.Main())
layer_tool = XCAFDoc_DocumentTool_LayerTool(doc.Main())

box = BRepPrimAPI_MakeBox(20., 40., 60.).Shape()  # Shape to add to document

main_assembly_label = shape_tool.NewShape()
#  main_assembly_label = shape_tool.AddShape(OCC_TopoDS.TopoDS_Compound())
sub_shape_label = shape_tool.AddSubShape(main_assembly_label, box)

step_writer = STEPCAFControl_Writer()

status = step_writer.Transfer(doc, STEPControl_AsIs)

if status != IFSelect_RetDone:
    raise AssertionError("data transfer failed")
status = step_writer.Write("test.stp")
if status != IFSelect_RetDone:
   raise Exception('write faild')
```

OkanPinar, 2020-11-17:
Firstly thanks for reply.

I've tried the code above both pythonocc-core version  7.4.0 and 7.4.1 in anaconda however result is the same.

tpaviot, 2020-11-18:
ok, thanks for the feedback. I ran the test using the current master branch (based on opencascade 7.5.0). Let me try with 7.4.0 and 7.4.1, this maybe something that ws fixed in the latest occt release.

tpaviot, 2020-12-01:
I checked, the code I suggest works using pythonocc-core version 7.4.1

closed 2020-12-01 by tpaviot