tpaviot/pythonocc-core #921: XCAFDoc_DocumentTool_LayerTool.SetLayer not working on compound

OkanPinar, 2020-11-16:
Hello everyone

Following code block create a simple step file with simple compound object. If entities will be added to document item by item layer information can be obtain on the step file like following.
[image]

However, after adding them into a compound object layers are not exported. Is this feature is not supported for compound objects?

```
        import OCC.Core.STEPCAFControl as OCC_STEP_XCAF_Control
        import OCC.Core.STEPControl as OCC_STEP_Control
        import OCC.Core.TDocStd as OCC_Doc_STD
        import OCC.Core.TCollection as OCC_Type_Collections
        import OCC.Core.XCAFDoc as OCC_XCaf_Doc
        import OCC.Core.BRepPrimAPI as OCC_Brep_Prim_API
        import OCC.Extend.DataExchange as OCC_Extend_Data_Exchange
        import OCC.Core.TDF as OCC_TDF
        import OCC.Core.TopoDS as OCC_TopoDS

        doc = OCC_Doc_STD.TDocStd_Document(OCC_Type_Collections.TCollection_ExtendedString("pythonocc-doc"))

        box_1 = OCC_Brep_Prim_API.BRepPrimAPI_MakeBox(20., 40., 60.).Shape()
        box_2 = OCC_Brep_Prim_API.BRepPrimAPI_MakeBox(40., 80., 120.).Shape()

        block = OCC_TopoDS.TopoDS_Compound()
        shape_handler = OCC_TopoDS.TopoDS_Builder()
        shape_handler.MakeCompound(block)
        shape_handler.Add(block, box_1)
        shape_handler.Add(block, box_2)

        shape_tool = OCC_XCaf_Doc.XCAFDoc_DocumentTool_ShapeTool(doc.Main())
        layer_tool = OCC_XCaf_Doc.XCAFDoc_DocumentTool_LayerTool(doc.Main())

        main_assembly_label = shape_tool.AddShape(block)

        layer_name = layer_tool.AddLayer(OCC_Type_Collections.TCollection_ExtendedString("test_layer"))

        layer_tool.SetLayer(main_assembly_label, layer_name)

        step_writer = OCC_STEP_XCAF_Control.STEPCAFControl_Writer()
        step_writer.SetLayerMode(True)

        status = step_writer.Transfer(doc, OCC_STEP_Control.STEPControl_AsIs)

        if status != OCC_Extend_Data_Exchange.IFSelect_RetDone:
            raise AssertionError("load failed")
        status = step_writer.Write("test.stp")
        if status != OCC_Extend_Data_Exchange.IFSelect_RetDone:
            raise Exception('Something bad happened')
```

tpaviot, 2020-11-17:
I confirm. However, can you please update your example with two shapes that does not self intersect (i.e. translate one of the shape far enough from the first one)

OkanPinar, 2020-11-17:
I've just revised my code with simple translation and module import modification. Result is the same.

```
    from OCC.Core.STEPCAFControl import STEPCAFControl_Writer
    from OCC.Core.STEPControl import STEPControl_AsIs
    from OCC.Core.TDocStd import TDocStd_Document
    from OCC.Core.TCollection import TCollection_ExtendedString
    from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool_ShapeTool, XCAFDoc_DocumentTool_LayerTool
    from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
    from OCC.Extend.DataExchange import IFSelect_RetDone
    from OCC.Core.TopoDS import TopoDS_Compound, TopoDS_Builder
    from OCC.Core.gp import gp_Trsf, gp_Pnt
    from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Transform

    doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))

    box_1 = BRepPrimAPI_MakeBox(20., 40., 60.).Shape()
    box_2 = BRepPrimAPI_MakeBox(40., 80., 120.).Shape()

    xform = gp_Trsf()
    xform.SetTranslation(gp_Pnt(0,0,0), gp_Pnt(40,0,0))
    brep_xform = BRepBuilderAPI_Transform(box_1, xform)
    brep_xform.Build()
    box_1_translated = brep_xform.Shape()

    block = TopoDS_Compound()
    shape_handler = TopoDS_Builder()
    shape_handler.MakeCompound(block)
    shape_handler.Add(block, box_1_translated)
    shape_handler.Add(block, box_2)

    shape_tool = XCAFDoc_DocumentTool_ShapeTool(doc.Main())
    layer_tool = XCAFDoc_DocumentTool_LayerTool(doc.Main())

    main_assembly_label = shape_tool.AddShape(block)

    layer_name = layer_tool.AddLayer(TCollection_ExtendedString("test_layer"))

    layer_tool.SetLayer(main_assembly_label, layer_name)

    step_writer = STEPCAFControl_Writer()
    step_writer.SetLayerMode(True)

    status = step_writer.Transfer(doc, STEPControl_AsIs)

    if status != IFSelect_RetDone:
        raise AssertionError("load failed")
    status = step_writer.Write("test.stp")
    if status != IFSelect_RetDone:
        raise Exception('Something bad happened')
```

tpaviot, 2020-11-18:
ok thank you for the report.

In the mean time, you may just have to avoid inserting compounds into the document, only shapes