tpaviot/pythonocc-core #923: How to get the handle of an AIS_PointCloud?

Pinija, 2020-11-18:
I have to Erase an AIS_Pointcloud from the display, with normal shapes (from a .stp file or created by the BrepPrimAPI) this can be accomplished by saving the aniobj handle which is the first return of the "DisplayShape" function and then erase it via context.Erase(handle).
However with the Pointcloud I can't use the DisplayShape function for it not being a TopoDS_Shape. When I display the pointcloud via display.context.Display(pointcloud, ...) there is no return such as an aniobj handle. The last Idea I have is to EraseAll and display everything I want keep again... Since this is far from an elegant solution (and then I would have to figure out how to get the coordinate system cross in the lower right corner to be displayed again), I would like to ask if there is a better way to accomplish what I'm trying.

tpaviot, 2020-12-01:
I don't understand your question, can you please elaborate and provide a small code that reproduces the issue

Pinija, 2020-12-01:
Thanks for your answer. 
This would be the minimal example of what I'm trying, this is working in until pythonocc 18.1, for older versions the GetHandle() functions have been removed I assume, I would like to rebiuld this for a newer OCC Version. 
________________________________________________________________
from OCC.gp import gp_Pnt
from OCC.Graphic3d import Graphic3d_ArrayOfPoints
from OCC.AIS import AIS_PointCloud

from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu  = init_display()

def pointcloud():

    points3d = Graphic3d_ArrayOfPoints(100000, True)
    
    for idx in range(100):
        for idy in range(100):
            for idz in range(10):
                x = 0 + idx * 0.1
                y = 0 + idy * 0.1
                z = 0 + idz * 0.1
                
                points3d.AddVertex(gp_Pnt(x, y, z))
    
    
    pointCloud = AIS_PointCloud()
    pointHandle = points3d.GetHandle()
    pointCloud.SetPoints(pointHandle)
    pointCloudHandle = pointCloud.GetHandle()
    display.Context.Display(pointCloudHandle)
    
    def ErasePointCloud():
        display.Context.Erase(pointCloudHandle)
        
    add_menu("Erase")
    add_function_to_menu("Erase", ErasePointCloud)
    display.FitAll()
    display.View_Iso()
    start_display()

if __name__ == "__main__":
    pointcloud()

___________________________________________________________
I can also specify the second part of my question further: I know how to display the "coordinate system cross (triedron)" via display.display_triedron(), when I try this function after Erasing the display (via EraseAll() ) this doesn't do anything. I think it is due to the "self.context.purgeDisplay()" method that has been added to EraseAll (in Display/OCCViewer.py) at sometime after 18.1, because when I comment the purge out in the sourcecode, the display_triedron works.

tpaviot, 2020-12-03:
I confirm that the ```PurgeDisplay``` method has to be removed

closed 2021-04-14 by tpaviot