tpaviot/pythonocc-core #925: StlAPI_Writer does not work

TobyLongley, 2020-11-26:
Hello guys, I'm a bit lost. I tried to use pythonOCC to bulk convert step files into stl files. Unfortunately I won't get any output. 
Here is my code:

```
from OCC.Core.STEPControl import STEPControl_Reader
from OCC.Core.StlAPI import StlAPI_Writer

input_file  = 'path_to_asm.stp'  
output_file = 'path_to_asm.stl'  

step_reader = STEPControl_Reader()
step_reader.ReadFile(input_file)
step_reader.TransferRoots()
myshape = step_reader.Shape()

stl_writer = StlAPI_Writer()
stl_writer.SetASCIIMode(True)
stl_writer.Write(myshape, output_file)
```

tpaviot, 2020-11-26:
The shape has to be explicitely meshed before being passed to the stl writer. You're trying to reinvent the wheel, consider using
```python
from OCC.Extend.DataExchange import read_step_file, write_stl_file
```

FYI see https://github.com/tpaviot/pythonocc-core/blob/7.4.1/src/Extend/DataExchange.py#L323

TobyLongley, 2020-11-26:
> The shape has to be explicitely meshed before being passed to the stl writer. You're trying to reinvent the wheel, consider using
>
> ```python
> from OCC.Extend.DataExchange import read_step_file, [...]

Thank you for your unbelivable fast reply :) I feel pretty bad for such an obvioiusly stupid question. I am pretty new to python etc. and get quite confused with the documentation for that library. 

I saw that I can read a step file with multiple files with ```def read_step_file_with_names_colors(filename):```  may I ask you how I access them? Or how to export the single parts as single stl files? e.g. I have an asselbmy of 4 parts. I read the assembly stl and save part1.stl, part2.stl...?

tpaviot, 2020-12-01:
I suggest you open another issue, it's another topic.

closed 2020-12-01 by tpaviot