tpaviot/pythonocc-core #928: Changing color of dimensions

StaffanHaglund, 2020-11-26:
Hi,
I display dimensions in my model but the default color green is hard to see clearly so I'd like to change it to red. This is where my headache starts. I have tried all I can think of, hopefully someone can help me out.
The code is shown below and it seems that all the different color settings are set to red but it is still green when shown. Note that "SetHilightColor" works to set the dimension color to red when highlighted but the it is still green when not highlighted. What am I missing?
Also, can I make the lines that the arrows are pointing at dashed somehow?

Any help is greatly appreciated
Regards
Staffan

**Code:** 
        dim=AIS_LengthDimension(e,pln)
        print 'dim.Color() before :',dim.Color(), Quantity_NOC_RED
        dim.SetColor(Quantity_NOC_RED)
        print 'dim.Color() after:',dim.Color(), Quantity_NOC_RED
        shp=dim.GetHandle()
        print 'display.Context.PreSelectionColor() before:',display.Context.PreSelectionColor(), Quantity_NOC_GREEN, Quantity_NOC_RED
        display.Context.SetPreselectionColor(Quantity_NOC_RED)
        display.Context.SetHilightColor(Quantity_NOC_RED) # This works for highlighting
        display.Context.SetSubIntensityColor(Quantity_NOC_RED)
        #display.Context.HilightWithColor(shp,Quantity_NOC_RED)
        print 'display.Context.PreSelectionColor() after:',display.Context.PreSelectionColor(), Quantity_NOC_GREEN, Quantity_NOC_RED
        display.Context.SetColor(shp, Quantity_NOC_RED, True)
        display.Context.Display(shp, True)
        print 'display.Context:',[x for x in dir(display.Context) if 'olor' in x]
        print 'dim:',[x for x in dir(dim) if 'olor' in x]

**Printout:**
dim.Color() before : 516 412
dim.Color() after: 412 412
display.Context.PreSelectionColor() before: 229 229 412
display.Context.PreSelectionColor() after: 412 229 412
display.Context: ['Color', 'DefaultColor', 'HasColor', 'HilightColor', 'HilightWithColor', 'PreSelectionColor', 'SelectionColor', 'SetColor', 'SetHilightColor', 'SetPreselectionColor', 'SetSubIntensityColor', 'SubIntensityColor', 'UnsetColor']
dim: ['Color', 'HasColor', 'HilightOwnerWithColor', 'SetColor', 'UnsetColor']:

[image]

tpaviot, 2020-12-01:
it could be easier to help you if you provide a complete script ready to run

StaffanHaglund, 2020-12-01:
Hi,
I have added a script that illustrates the issue:

    from OCC.AIS import AIS_LengthDimension,AIS_TextLabel
    from OCC.Display.SimpleGui import init_display
    from OCC.Display.OCCViewer import rgb_color
    from OCC.TopoDS import topods_Face,topods_Edge,topods_Vertex,topods
    from OCC.gp import gp_Pnt,gp_Dir,gp_Pln, gp_Vec, gp_Pnt2d, gp_Trsf, gp_Ax2, gp_DZ,gp_DX, gp_Elips, gp_Circ
    from core_geometry_utils import make_edge, make_vertex, make_wire, make_face,make_extrusion,make_n_sided, translate_shp,rotate_shp_3_axis
    from core_topology_traverse import Topo
    import numpy as np
    display, start_display, add_menu, add_function_to_menu, qtViewer3d = init_display()
    from OCC.BRepBuilderAPI import BRepBuilderAPI_Copy, BRepBuilderAPI_Transform, BRepBuilderAPI_MakeSolid, BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeFace
    from OCC.Quantity import Quantity_Color,Quantity_NOC_BROWN1,Quantity_TOC_RGB, Quantity_NOC_ALICEBLUE,Quantity_NOC_WHITE,Quantity_NOC_BLACK,Quantity_NOC_RED,Quantity_NOC_RED1,Quantity_NOC_GREEN
    from OCC.BRepPrimAPI import BRepPrimAPI_MakePrism, BRepPrimAPI_MakeCylinder, BRepPrimAPI_MakeBox, BRepPrimAPI_MakeCone
    from OCC.TopAbs import TopAbs_EDGE, TopAbs_FACE, TopAbs_SHAPE
    from TopologyUtils import is_edge, is_face, is_solid,is_vertex,TopologyExplorer
    from OCC.TopExp import TopExp_Explorer
    from OCC.TCollection import TCollection_AsciiString,TCollection_ExtendedString 
    
    def extract_edges(bodies):
        edges=[]
        if not isinstance(bodies, list):
            bodies=[bodies]
        for body in bodies:
            anEdgeExplorer = TopExp_Explorer(body, TopAbs_EDGE)
            while anEdgeExplorer.More():
                anEdge = topods.Edge(anEdgeExplorer.Current())
                edges.append(anEdge)
                anEdgeExplorer.Next()    
        return edges
    
    def main():
        S=BRepPrimAPI_MakeBox(gp_Pnt(0,0,0), 100, 100, 100).Solid()
        edges=extract_edges(S)
    
        #add dimension to edge    
        pln = gp_Pln(gp_Pnt(0,50,0),gp_Dir(1,0,0))
        dim=AIS_LengthDimension(edges[0], pln)
        dim.SetColor(Quantity_NOC_RED)
    
        #-------------------
        # I have tried different settings......
        #-----------------
        if 0:
            print 'dim.Color() before :',dim.Color(), Quantity_NOC_RED
            dim.SetColor(Quantity_NOC_RED)
            print 'dim.Color() after:',dim.Color(), Quantity_NOC_RED
            shp=dim.GetHandle()
            print 'display.Context.PreSelectionColor() before:',display.Context.PreSelectionColor(), Quantity_NOC_GREEN, Quantity_NOC_RED
            display.Context.SetPreselectionColor(Quantity_NOC_RED)
            display.Context.SetHilightColor(Quantity_NOC_RED) # This works for highlighting
            display.Context.SetSubIntensityColor(Quantity_NOC_RED)
            display.Context.HilightWithColor(shp,Quantity_NOC_RED)
            print 'display.Context.PreSelectionColor() after:',display.Context.PreSelectionColor(), Quantity_NOC_GREEN, Quantity_NOC_RED
            display.Context.SetColor(shp, Quantity_NOC_RED, False)
        
        shp=dim.GetHandle()
        display.Context.Display(shp, True)    
        
        #adding a text
        t=AIS_TextLabel()
        t.SetPosition(gp_Pnt(0,-50,50))
        t.SetText(TCollection_ExtendedString('(this text is red but not the dimension)'))
        t.SetColor(Quantity_NOC_RED)
        display.Context.Display(t.GetHandle(), True)
        display.DisplayShape(S)
    
    if __name__=="__main__":
        main()
        display.FitAll()
        start_display()

tpaviot, 2020-12-02:
You're right, using the green (default) color for dimension is not visible:
[image]

Try to use the ```Prs3d_DimensionAspect```, for example:
```python
from OCC.Core.gp import gp_Dir, gp_Ax2, gp_Circ, gp_Pnt
from OCC.Core.AIS import AIS_Shape, AIS_RadiusDimension
from OCC.Core.Quantity import Quantity_Color, Quantity_NOC_BLACK
from OCC.Core.Prs3d import Prs3d_DimensionAspect
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()

c = gp_Circ(gp_Ax2(gp_Pnt(200., 200., 0.), gp_Dir(0., 0., 1.)), 80)
ec = BRepBuilderAPI_MakeEdge(c).Edge()
ais_shp = AIS_Shape(ec)
display.Context.Display(ais_shp, True)

rd = AIS_RadiusDimension(ec)
the_aspect = Prs3d_DimensionAspect()
the_aspect.SetCommonColor(Quantity_Color(Quantity_NOC_BLACK))
rd.SetDimensionAspect(the_aspect)

display.Context.Display(rd, True)
display.FitAll()
start_display()
```
and you should get something more readable
[image]

If it's ok for yu, please close the issue. I added this code to the example suite.

StaffanHaglund, 2020-12-02:
Thank you, that helped.

closed 2020-12-02 by StaffanHaglund

abhi1p, 2022-11-06:
Hi,
I'm facing same issue with Prs3d_DimensionAspect

`from OCC.Core.GC import GC_MakePlane
from OCC.Core.Prs3d import Prs3d_DimensionAspect
from OCC.Core.PrsDim import PrsDim_LengthDimension
from OCC.Core.Quantity import Quantity_Color, Quantity_NOC_BLACK
from OCC.Core.gp import gp_Pnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire, BRepBuilderAPI_MakeFace
from OCC.Display.SimpleGui import init_display

p1 = gp_Pnt(0, 0, 0)  # point 1
p2 = gp_Pnt(0, 50, 0)  # point 2
p3 = gp_Pnt(50, 50, 0)  # point 3
p4 = gp_Pnt(50, 0, 0)  # point 4

edge1 = BRepBuilderAPI_MakeEdge(p1, p2).Edge()  # edge from point 1 to point 2
edge2 = BRepBuilderAPI_MakeEdge(p2, p3).Edge()  # edge from point 2 to point 3
edge3 = BRepBuilderAPI_MakeEdge(p3, p4).Edge()  # edge from point 3 to point 4
edge4 = BRepBuilderAPI_MakeEdge(p4, p1).Edge()  # edge from point 4 to point 1

wire = BRepBuilderAPI_MakeWire(edge1, edge2, edge3, edge4).Wire()  # make wire from edges
face = BRepBuilderAPI_MakeFace(wire).Face()  # make face from wire

display, start_display, add_menu, add_function_to_menu = init_display()  # init screen for display
display.DisplayShape(face, update=True)  # display rectangle

plane = GC_MakePlane(p1, p2, p3).Value()

anAspect = Prs3d_DimensionAspect()
anAspect.MakeArrows3d(False)
anAspect.MakeText3d(False)
anAspect.MakeTextShaded(True)
anAspect.SetCommonColor(Quantity_Color(Quantity_NOC_BLACK))

aLenDim = PrsDim_LengthDimension(p1, p2, plane.Pln())
aLenDim.SetAspect(anAspect)
display.Context.Display(aLenDim,True)

start_display()  # start screen loop`