tpaviot/pythonocc-core #937: Are there any polygon morphology operations I'm missing?

inventor2525, 2020-12-04:
Such as OpenCV does [here](https://docs.opencv.org/master/d9/d61/tutorial_py_morphological_ops.html)

I'm looking mostly for erosion and dilation of OCC faces.

Currently I'm using [Shapely](https://pypi.org/project/Shapely/) to (they call their function) "buffer" a polygon, and then convert it to an OCC face and extrude my 3d geometry up from there, but Shapely builds it's polygons of solely line segments and I'd like to use the full descriptive power of a b rep (eg, actual arcs for end caps).

Closest I've seen is Geom2d_OffsetCurve, but there's be a pretty bit to build on top for that.

tpaviot, 2020-12-04:
I'm not sure to fully understand what you need to finally achieve, you may check https://github.com/mathLab/PyGeM

inventor2525, 2020-12-04:
PyGeM's probably a bit much. I'm immagining a single occ function I've yet to find (if there is one), but vocab changes slightly in differnt contexts.

The simplest way I could think to explain this, is roughly:
`
def buffer(geometry :Union[Point, Edge, Face], ammount :float):
  if isinstance(geometry, Point):
    #make circle centered on point, ammound in radius
`

inventor2525, 2020-12-04:
PyGeM's probably? a bit much.

I'm imagining a single occ function I've yet to find (if there is one), but vocab changes slightly in different contexts.

I think sudo code [here](https://gist.github.com/inventor2525/0def58e45601c58746b1f8b2d2943b32#file-buffer_sudo_example-py) may clear this this up more completely, but the picture [here](https://warehouse-camo.ingress.cmh1.psfhosted.org/133147f8ffe1a9ac1e51fad6ce7ac138b9a2ddcf/68747470733a2f2f63322e737461746963666c69636b722e636f6d2f362f353536302f33313330313739303038365f623334373265613465395f632e6a7067) really helps more maybe once your close. The left side of the image shows the effect of buffering a linestripe (akin to a wire -- shown there in dark blue), and the light blue polygon that results. The right, what happens when you buffer a polygon by a negative value (grey being input, blue being result). -- If you continued the first, holes would disappear, if you continued the second, eventually the polygon would disappear.

Appologies for the accidental early post, and thank you for this awesome framework and support!

tpaviot, 2020-12-04:
Ok, I got it. OCC provides good 2d primitives and algorithms (offset, trimming, boolean ops), it can certainly be done. It requires quite a piece of work though.

inventor2525, 2020-12-04:
Gotcha. Ok, well, if I end up making it I'll go ahead and post it, but I may be able to do what I need for now in other ways easier.
Thank you!

closed 2020-12-04 by inventor2525