tpaviot/pythonocc-core #942: unable to compile v7.4.x according to the directions

salfter, 2020-12-08:
I have a Gentoo ebuild that worked for v0.18.1, but it appears to not work with newer versions.  It appears to not build a bunch of header files with SWIG, then complains that those header files are missing:  

```
salfter@janeway /tmp/pythonocc-core/build $ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
--
-- Found OpenGL: /usr/lib64/libOpenGL.so
-- Found Python3: /usr/bin/python3.9 (found version "3.9.0") found components: Interpreter Development
-- Python3 interpreter:/usr/bin/python3.9
-- Python include directory:
-- Python library release: /usr/lib64/libpython3.9.so
-- Found SWIG: /usr/bin/swig (found suitable version "4.0.2", minimum required is "3.0.11")
-- Disabled SWIG warnings
-- Disabled deprecation warnings for oce
-- OpenCASCADE found.
-- pythonOCC modules will be installed to: /usr/lib/python3.9/site-packages/OCC
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/pythonocc-core/build
salfter@janeway /tmp/pythonocc-core/build $ make
Scanning dependencies of target XmlTObjDrivers_swig_compilation
[  0%] Swig compile src/SWIG_files/wrapper/XmlTObjDrivers.i for python
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:57: Error: Unable to find 'NCollection_TypeDef.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:58: Error: Unable to find 'NCollection_Array1.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:59: Error: Unable to find 'NCollection_Array2.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:60: Error: Unable to find 'NCollection_Map.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:61: Error: Unable to find 'NCollection_DefaultHasher.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:62: Error: Unable to find 'NCollection_List.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:63: Error: Unable to find 'NCollection_Sequence.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:64: Error: Unable to find 'NCollection_DataMap.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:65: Error: Unable to find 'NCollection_IndexedMap.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:66: Error: Unable to find 'NCollection_IndexedDataMap.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:67: Error: Unable to find 'NCollection_DoubleMap.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:68: Error: Unable to find 'NCollection_DefineAlloc.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:69: Error: Unable to find 'Standard_Macro.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:70: Error: Unable to find 'Standard_DefineAlloc.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:71: Error: Unable to find 'NCollection_UBTree.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:72: Error: Unable to find 'NCollection_UBTreeFiller.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:73: Error: Unable to find 'NCollection_Lerp.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:74: Error: Unable to find 'NCollection_Vector.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:75: Error: Unable to find 'NCollection_Vec2.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:76: Error: Unable to find 'NCollection_Vec3.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:77: Error: Unable to find 'NCollection_Vec4.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:78: Error: Unable to find 'NCollection_Mat4.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:79: Error: Unable to find 'NCollection_TListIterator.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:80: Error: Unable to find 'NCollection_UtfString.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:81: Error: Unable to find 'NCollection_UtfIterator.hxx'
/tmp/pythonocc-core/src/SWIG_files/wrapper/NCollection.i:82: Error: Unable to find 'NCollection_SparseArray.hxx'
make[2]: *** [CMakeFiles/XmlTObjDrivers_swig_compilation.dir/build.make:84: CMakeFiles/_XmlTObjDrivers.dir/XmlTObjDriversPYTHON.stamp] Error 1
make[2]: *** Deleting file 'CMakeFiles/_XmlTObjDrivers.dir/XmlTObjDriversPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:1021: CMakeFiles/XmlTObjDrivers_swig_compilation.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
```

tpaviot, 2020-12-08:
opencascade should be found by cmake, but if not (what is actually your case), you have to pass ```-DOCE_INCLUDE_PATH=/occt_include_dir``` to cmake

tpaviot, 2020-12-08:
No I'm wrong, opencascade is found. Could you please ```make VERBOSE=1``` ?

salfter, 2020-12-08:
```
salfter@janeway /tmp/pythonocc-core-7.4.1/build $ make VERBOSE=1
/usr/bin/cmake -S/tmp/pythonocc-core-7.4.1 -B/tmp/pythonocc-core-7.4.1/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /tmp/pythonocc-core-7.4.1/build/CMakeFiles /tmp/pythonocc-core-7.4.1/build/CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/tmp/pythonocc-core-7.4.1/build'
make  -f CMakeFiles/XmlTObjDrivers_swig_compilation.dir/build.make CMakeFiles/XmlTObjDrivers_swig_compilation.dir/depend
make[2]: Entering directory '/tmp/pythonocc-core-7.4.1/build'
cd /tmp/pythonocc-core-7.4.1/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/pythonocc-core-7.4.1 /tmp/pythonocc-core-7.4.1 /tmp/pythonocc-core-7.4.1/build /tmp/pythonocc-core-7.4.1/build /tmp/pythonocc-core-7.4.1/build/CMakeFiles/XmlTObjDrivers_swig_compilation.dir/DependInfo.cmake --color=
Dependee "/tmp/pythonocc-core-7.4.1/build/CMakeFiles/XmlTObjDrivers_swig_compilation.dir/DependInfo.cmake" is newer than depender "/tmp/pythonocc-core-7.4.1/build/CMakeFiles/XmlTObjDrivers_swig_compilation.dir/depend.internal".
Dependee "/tmp/pythonocc-core-7.4.1/build/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/tmp/pythonocc-core-7.4.1/build/CMakeFiles/XmlTObjDrivers_swig_compilation.dir/depend.internal".
Scanning dependencies of target XmlTObjDrivers_swig_compilation
make[2]: Leaving directory '/tmp/pythonocc-core-7.4.1/build'
make  -f CMakeFiles/XmlTObjDrivers_swig_compilation.dir/build.make CMakeFiles/XmlTObjDrivers_swig_compilation.dir/build
make[2]: Entering directory '/tmp/pythonocc-core-7.4.1/build'
[  0%] Swig compile src/SWIG_files/wrapper/XmlTObjDrivers.i for python
/usr/bin/cmake -E make_directory Unix/x86_64-Release- Unix/x86_64-Release-
/usr/bin/cmake -E touch /tmp/pythonocc-core-7.4.1/build/CMakeFiles/_XmlTObjDrivers.dir/XmlTObjDriversPYTHON.stamp
/usr/bin/cmake -E env SWIG_LIB=/usr/share/swig/4.0.2 /usr/bin/swig -python -fvirtual -py3 -w302,401,402,412,314,509,512,504,325,503,520,350,351,383,389,394,395,404 -DCSFDB -DHAVE_CONFIG_H -outdir Unix/x86_64-Release- -c++ -interface _XmlTObjDrivers -I/tmp/pythonocc-core-7.4.1 -I/tmp/pythonocc-core-7.4.1/OPENGL_INCLUDE_DIR -I/usr/include/opencascade -I/tmp/pythonocc-core-7.4.1/src/SWIG_files/headers -I/tmp/pythonocc-core-7.4.1/src/Tesselator -I/tmp/pythonocc-core-7.4.1/src/Visualization -o Unix/x86_64-Release-/XmlTObjDriversPYTHON_wrap.cxx /tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/XmlTObjDrivers.i
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:57: Error: Unable to find 'NCollection_TypeDef.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:58: Error: Unable to find 'NCollection_Array1.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:59: Error: Unable to find 'NCollection_Array2.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:60: Error: Unable to find 'NCollection_Map.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:61: Error: Unable to find 'NCollection_DefaultHasher.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:62: Error: Unable to find 'NCollection_List.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:63: Error: Unable to find 'NCollection_Sequence.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:64: Error: Unable to find 'NCollection_DataMap.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:65: Error: Unable to find 'NCollection_IndexedMap.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:66: Error: Unable to find 'NCollection_IndexedDataMap.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:67: Error: Unable to find 'NCollection_DoubleMap.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:68: Error: Unable to find 'NCollection_DefineAlloc.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:69: Error: Unable to find 'Standard_Macro.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:70: Error: Unable to find 'Standard_DefineAlloc.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:71: Error: Unable to find 'NCollection_UBTree.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:72: Error: Unable to find 'NCollection_UBTreeFiller.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:73: Error: Unable to find 'NCollection_Lerp.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:74: Error: Unable to find 'NCollection_Vector.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:75: Error: Unable to find 'NCollection_Vec2.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:76: Error: Unable to find 'NCollection_Vec3.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:77: Error: Unable to find 'NCollection_Vec4.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:78: Error: Unable to find 'NCollection_Mat4.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:79: Error: Unable to find 'NCollection_TListIterator.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:80: Error: Unable to find 'NCollection_UtfString.hxx'
/tmp/pythonocc-core-7.4.1/src/SWIG_files/wrapper/NCollection.i:81: Error: Unable to find 'NCollection_UtfIterator.hxx'
[... cut ...] .hxx'
make[2]: *** [CMakeFiles/XmlTObjDrivers_swig_compilation.dir/build.make:84: CMakeFiles/_XmlTObjDrivers.dir/XmlTObjDriversPYTHON.stamp] Error 1
make[2]: *** Deleting file 'CMakeFiles/_XmlTObjDrivers.dir/XmlTObjDriversPYTHON.stamp'
make[2]: Leaving directory '/tmp/pythonocc-core-7.4.1/build'
make[1]: *** [CMakeFiles/Makefile2:1021: CMakeFiles/XmlTObjDrivers_swig_compilation.dir/all] Error 2
make[1]: Leaving directory '/tmp/pythonocc-core-7.4.1/build'
make: *** [Makefile:150: all] Error 2
```

Dr-Terrible, 2021-01-21:
For future reference: this is not a `pythonocc` bug, but a bug of CMake's macro `OpenCASCADEConfig.cmake` shipped with package `sci-libs/opencascade-7.4.0` by distro Gentoo. Here is Gentoo's [bug report](https://bugs.gentoo.org/763162), and here is OpenCASCADE's [bug report](https://tracker.dev.opencascade.org/view.php?id=32035).

TL;DR: the aforementioned find macro ignores the `CASROOT` environment variable, thus erroneously assumes that the include directory is located at `/usr/include/opencascade` instead of `/usr/lib64/opencascade-7.4.0/ros/include/opencascade`.

As a temporarily solution for Gentoo users, you need to add the [patch](https://763162.bugs.gentoo.org/attachment.cgi?id=680725) proposed by Fabio Rossi to `/etc/portage/patches/sci-libs/opencascade/`, recompile `sci-libs/opencascade-7.4.0` to apply the new patch, and then `pythonocc-7.4.1` should be able to compile successfully.

For simplicity, here is my /etc/portage/patches/sci-libs/opencascade/opencascade-7.4.0-fix-cmake-include-path.patch
```patch
--- a/adm/templates/OpenCASCADEConfig.cmake.in
+++ b/adm/templates/OpenCASCADEConfig.cmake.in
@@ -16,19 +16,7 @@
 set (OpenCASCADE_MAINTENANCE_VERSION "@OCC_VERSION_MAINTENANCE@")
 set (OpenCASCADE_DEVELOPMENT_VERSION "@OCC_VERSION_DEVELOPMENT@")
 
-# Compute the installation prefix from this OpenCASCADEConfig.cmake file 
-# location, by going up one level + one level if "cmake" + one level if "lib".
-# This is made to support different locations of CMake files:
-# - in UNIX style: $INSTALL_DIR/lib/cmake/opencascade-<version>
-# - in Windows style: $INSTALL_DIR/cmake
-get_filename_component (OpenCASCADE_INSTALL_PREFIX "${CMAKE_CURRENT_LIST_FILE}" PATH)
-get_filename_component (OpenCASCADE_INSTALL_PREFIX "${OpenCASCADE_INSTALL_PREFIX}" PATH)
-if (OpenCASCADE_INSTALL_PREFIX MATCHES "/cmake$")
-  get_filename_component (OpenCASCADE_INSTALL_PREFIX "${OpenCASCADE_INSTALL_PREFIX}" PATH)
-endif()
-if (OpenCASCADE_INSTALL_PREFIX MATCHES "/lib$")
-  get_filename_component (OpenCASCADE_INSTALL_PREFIX "${OpenCASCADE_INSTALL_PREFIX}" PATH)
-endif()
+set (OpenCASCADE_INSTALL_PREFIX "$ENV{CASROOT}")
 
 # Set OpenCASCADE paths to headers, binaries, libraries, resources, tests, samples, data
 set (OpenCASCADE_BINARY_DIR   "${OpenCASCADE_INSTALL_PREFIX}/@INSTALL_DIR_BIN@")
```

You should be able to see the new patch applied in the output of emerge:
```sh
>>> Preparing source in /var/tmp/portage/sci-libs/opencascade-7.4.0-r3/work/occt-V7_4_0 ...
 * Applying opencascade-7.4.0-find-qt.patch ...                                                                                            [ ok ]
 * Applying opencascade-7.4.0-fix-install.patch ...                                                                                        [ ok ]
 * Applying opencascade-7.4.0-fix-cmake-include-path.patch ...                                                                             [ ok ]
 * User patches applied.
```

With this solution I was able to compile and install CadQuery-2.0 on Gentoo, which requires pythonocc-7.4.1 as a dependency.

salfter, 2021-01-21:
> As a temporarily solution for Gentoo users, you need to add the patch proposed by Fabio Rossi to /etc/portage/patches/sci-libs/opencascade/...

Did that, and the latest CadQuery installed with the ebuild I'd written a while back.  All that's left is to test it. :)

closed 2021-01-21 by salfter