tpaviot/pythonocc-core #947: How to use occ to generate a cross-sectional view and convert it into a two-dimensional engineering drawing for display, I did not find any relevant sample programs in the occ sample programs？

luoyu-123, 2020-12-14:
I want to use occ to implement the cross-sectional view function of a three-dimensional object, and mark the data on the cross-section, the effect is the same as the picture below, but I did not find the relevant sample program？
thank you very much！！
best wishes
tpaviot

[image]

tpaviot, 2020-12-14:
You can combine https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_clip_planes.py and https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_dimensions.py

OkanPinar, 2020-12-16:
For creating 2D drawing from 3D object  HLRBRep_PolyAlgo and BRepAlgoAPI_Section classes can be useful also.

closed 2021-01-20 by luoyu-123

Bill-XU, 2024-09-13:
> thank you

@luoyu-123 
Hello, have you implemented "mark the data on the cross-section"?
Curious about how to determine the edges to measure and mark the data.

Best regards,
Bill