tpaviot/pythonocc-core #951: A Shape and a Plane intersection polygon

Vahid44, 2020-12-25:
Hello everyone, 
I got the intersection between a shape and a plane using "BRepAlgoAPI.BRepAlgoAPI_Section" function.
Is it possible to extract a polygon from the intersection? (Or the points of the intersection)

This is my code to extract the intersection. and the image is the result.
```
plan = gp_Pln(gp_Pnt(0,0,4), gp_Dir(0,0,1))

section_face = OCC.Core.BRepBuilderAPI.BRepBuilderAPI_MakeFace(plan, -10, 10, -10, 10).Face()

settings = ifcopenshell.geom.settings()
settings.set(settings.USE_PYTHON_OPENCASCADE, True)
shape = ifcopenshell.geom.create_shape(settings, product).geometry

section = OCC.Core.BRepAlgoAPI.BRepAlgoAPI_Section(section_face, shape).Shape()

```
[image]

Sorry if it is a little trivial, but I did not find anything to solve my problem.
Best regards.

OkanPinar, 2020-12-26:
You can use TopologyExplorer to obtain vertices and edges.

Vahid44, 2020-12-29:
Thank you very much, it totally solved my problem. 
I first extract the edges. And then from the edges construct a polygon.
Is there any comprehensive documentation for pythonocc? Since I searched for a while but could not find anything.

OkanPinar, 2020-12-29:
Spending some time on @tpaviot's utils (https://github.com/tpaviot/pythonocc-utils) repository and opencascade forum(https://old.opencascade.com/forums) can be good for obtain main ideas.

OkanPinar, 2021-01-10:
The following site is great to study some code related with opencascade. 

http://www.creativecadtechnology.com/OCC/ShowScript?userName=learnMoreAboutOCC&groupName=demo&scriptName=aaabbbccc

Vahid44, 2021-01-11:
Thank you very much, dear OkanPinar.

AnganMitra, 2021-03-01:
@Vahid44 can you share your solution snippet to first extract the edges and then the polygon.? I was trying with the following snippet but with no luck `section =              OCC.Core.BRepAlgoAPI.BRepAlgoAPI_Section(section_face, shape).Shape()
section_edges = list(OCCUtils.Topo(section).edges())
ec += len(section_edges)
print ("ec----------",ec)
if len(section_edges) > 0:
              edges = OCC.Core.TopTools.TopTools_HSequenceOfShape()
              for edge in section_edges:
                  edges.Append(edge)
              edges_handle = OCC.Core.TopTools.TopTools_HSequenceOfShape(edges)

              wires = OCC.Core.TopTools.TopTools_HSequenceOfShape()
              wires_handle = OCC.Core.TopTools.TopTools_HSequenceOfShape(wires)
              
              OCC.Core.ShapeAnalysis.ShapeAnalysis_FreeBounds_ConnectEdgesToWires(edges_handle, 1e-5, True, wires_handle)
              # wires = wires_handle.GetObject()
              wires = wires_handle
              for i in range(wires.Length()):
                  wire_shape = wires.Value(i+1)
                  wire = topods.Wire(wire_shape)`

Vahid44, 2021-03-05:
Hello @AnganMitra, and sorry for the late reply.
The following code extracts the edges of the intersection of a plane and a shape.
It extracts the edges and the points of the edges, but the polygon extraction part behaves correctly just for simple shapes.

```
def Space_Section (IfcSpace, Elevation):      # Intersects a plane and a space in a certain elevation and returns the result as a List of ordered points
    if IfcSpace.is_a('IfcSpace'):
        aShape = IfcObj_to_TopoDsObj(IfcSpace)
        
        ## middel of the Space coordination (zMidP), type: number
        xMidP = get_aligned_boundingbox(aShape)[0].Coord()[0]
        yMidP = get_aligned_boundingbox(aShape)[0].Coord()[1]
        
        # defining a (Horizontal) plane to intersect with the space shape. (z = "Elevation")
        plane = gp_Pln(gp_Pnt(xMidP,yMidP,Elevation), gp_Dir(0,0,1))

        # intersect the plane and the space:
        section_face = OCC.Core.BRepBuilderAPI.BRepBuilderAPI_MakeFace(plane, -150, 150, -150, 150).Face()
        section = OCC.Core.BRepAlgoAPI.BRepAlgoAPI_Section(section_face, aShape).Shape()

        # get the edges and vertices of the intersection:
        Unordered_Edge_List = []
        Ordered_Vertex_List = []

        edges = list(TopologyExplorer(section).edges())
        for i in range(len(edges)):
            firstVert = OCC.Core.TopExp.topexp.FirstVertex(edges[i])
            p1_edge = BRep_Tool.Pnt(firstVert).Coord()
            p1_edge = tuple(map(lambda x: isinstance(x, float) and round(x, 3) or x, p1_edge))
            lastVert = OCC.Core.TopExp.topexp.LastVertex(edges[i])
            p2_edge = BRep_Tool.Pnt(lastVert).Coord()
            p2_edge = tuple(map(lambda x: isinstance(x, float) and round(x, 3) or x, p2_edge))
            TheEdge = [p1_edge, p2_edge]
            Unordered_Edge_List.append(TheEdge)

        if Unordered_Edge_List != []:
            Ordered_Vertex_List.append(Unordered_Edge_List[0][1])

        # order the points (to extract a polygon from ordered points)
        while Unordered_Edge_List:
            for x in Unordered_Edge_List:
                if Ordered_Vertex_List[-1] == x[0]:
                    Ordered_Vertex_List.append(x[1])
                    Unordered_Edge_List.remove(x)

         #Space_polygon = Polygon(Ordered_Vertex_List)
        return Ordered_Vertex_List

    else:
        return ('It is not an IfcSpace!')
```
      

Feel free if you had any questions or suggestions ;)