tpaviot/pythonocc-core #954: Installing OCC offline

syigit61, 2021-01-14:
I need to use BladeX in offline computer. Is it possible to install OCC module offline dependency ? I would be grateful if you could help me.

OkanPinar, 2021-01-14:
You can fallow the steps that are described in following link.

https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md

tpaviot, 2021-01-30:
Can you please elaborate on what you mean with "offline installation" ?

syigit61, 2021-01-30:
OCC ask many dependencies, my question is on is there any way to install OCC on computer without internet ?

tpaviot, 2021-02-02:
It's not a pythonocc related question I guess. Rather something related to conda. Without additional information, I suggest to create a conda installation on a first machine, then simply copy/paste the environment to the other machine.

closed 2021-02-02 by tpaviot